remap-keys/remap #4: 基本Viewレイアウトを作成する

adamrocker, 2020-12-15:
**機能**

- CONFIGURE：キーボードの設定（メイン画面）
- KEY TESTER：キーボードのキーが生きているかを確認（別サイトに飛ばす）
- DESIGN：キーボードレイアウトJSONを簡易的に確認するツール
- SETTINGS：本アプリの設定

**CONFIGURE**

- top
    - JSONを読み込み
    - LAYOUTS切り替え
- side
    - keymap
    - basic
    - macros
    - lighting
- footer
    - SAVEでJSONを保存

yoichiro, 2020-12-15:
既存のVIAアプリのConfigure画面は「キーボードが認識されないと何も表示されない」ようになっていて、さらに「Import Keymap」でキーボードのレイアウトが書かれたJSONファイルを読み込んだとしても、対応するキーボードが物理的に接続されないと何も表示されません。この動作は個人的には不親切かな、と思っていて、VIA対応の（正確にはJSONファイルを本プロダクトが持っている）キーボードを選択肢さえすれば、物理的にそれが接続されていなくても、キーレイアウト（キーコードの割り当ては空白）が表示されていいんじゃないか、と。そのうえで「未接続」とわかれば良いのかな、と思います。

そして、接続されていなくてもキーレイアウトが表示される挙動であれば、DESIGN画面は必要なくなるのかな、と思いますが、どうでしょう？

adamrocker, 2020-12-16:
良いですね。未接続である事を明示して、それによりどういう影響があるかを分かるようにしてあげることで「設定したのに反映されない」という混乱・不安を解消できそうですね。

あとURLでキーボードの設定画面に飛べるようにしてあげると、キーボード作者がURLを貼るだけで利用者にキーカスタマイズを促せて便利かなと思いますがいかがでしょうか？キーボードが接続されていなくても画面が表示できる上記の改善とBest Muchな気がします。

例えば、
/configure/lunakey_mini/22868/1 にアクセスするとLunakey Miniのキーカスタマイズ画面を表示できるという感じです。

yoichiro, 2020-12-16:
> あとURLでキーボードの設定画面に飛べるようにしてあげると、キーボード作者がURLを貼るだけで利用者にキーカスタマイズを促せて便利かなと思いますがいかがでしょうか？キーボードが接続されていなくても画面が表示できる上記の改善とBest Muchな気がします。

確かに！キーボードごとのPermalink、良さそうです！

closed 2020-12-20 by adamrocker