remap-keys/remap #30: UI Flow

yoichiro, 2020-12-24:
# 制約条件

* キーボードを接続していない状態では、キーマップの編集はできない。
* ユーザ未認可のキーボードは、接続した状態で認可をさせないと扱うことができない。
* 認可済みキーボードは、接続されていればすぐに扱うことができるが、複数存在する可能性がある。
* 同じVendor ID、Product IDのキーボードを複数接続した際の動作はサポートしない。
* 編集対象のキーボードは同時に1つだけ。

# 存在しうる状態

o. 起動時
a. キーボードが1台も接続されていない。
b. 未認可キーボードが接続されている。
c. 認可済みキーボードが1台接続されている。
d. 認可済みキーボードと未認可キーボードが同時に接続されている。
e. 認可済みキーボードが2台以上接続されている。

# できること

* 認可済みキーボードの一覧取得。
* 認可済みキーボードの接続/切断イベントの受信。

# 状態遷移

1. oの状態で、認可済みキーボードが1台接続されていると、cの状態になる。
2. oの状態で、認可済みキーボードが2台以上接続されていると、eの状態になる。
3. oの状態で、認可済みキーボードがないと、aの状態になる。
4. aの状態で、未認可キーボードを接続すると、bの状態になる。
5. aの状態で、認可済みキーボードが接続されると、cの状態になる。
6. bの状態で、ユーザが未認可キーボードを認可すると、cの状態になる。
7. bの状態で、認可済みキーボードが接続されると、cの状態になる。
8. cの状態で、未認可キーボードを接続すると、dの状態になる。
9. cの状態で、認可済みキーボードが接続されると、eの状態になる。
10. dの状態で、ユーザが未認可キーボードを認可すると、eの状態になる。
11. （eの状態では認可済みキーボードのいずれかを編集している）

# UI上で起きること

1. 認可済みキーボードの一覧取得を行って、結果が1件の場合は、そのキーボードの編集を開始する。
2. 認可済みキーボードの一覧取得を行って、結果が2件以上の場合は、どのキーボードを編集するかをユーザに選択させて、その結果のキーボードの編集を開始する。
3. 認可済みキーボードの一覧取得を行って、結果が0件の場合は、特に何もしない。
4. 未認可キーボードは検出されないので、特にやることなし。
5. 認可済みキーボードの接続イベントを捕捉して、そのキーボードの編集を開始する。
6. ユーザが「新規キーボードの接続」ボタンを押して未認可キーボードの認可をChromeに要求する。認可後に、そのキーボードの編集を開始する。
7. 認可済みキーボードの接続イベントを捕捉して、そのキーボードの編集を開始する。
8. 未認可キーボードは検出されないので、特にやることなし。
9. 認可済みキーボードの接続イベントを捕捉して、そのキーボードの編集に切り替えるかどうかユーザに問い合わせる。Yesの場合はそのキーボードの編集を開始し、Noの場合は何もしない。
10. ユーザが「新規キーボードの接続」ボタンを押して未認可キーボードの認可をChromeに要求する。認可後に、そのキーボードの編集に切り替えるかどうかユーザに問い合わせる。Yesの場合はそのキーボードの編集を開始し、Noの場合は何もしない。
11. 編集中ではない別の認可済みキーボードをユーザが選択して、そのキーボードの編集を開始する。

# 必要となるUI

* 新規キーボードの認可要求ボタン。
* 接続された認可済みキーボードの一覧。
* 現在編集中のキーボードの表示。
* 別の認可済みキーボードに編集を切り替えるUI。
* 別の認可済みキーボードに編集を切り替えるかどうか問うUI。

yoichiro, 2020-12-24:
@adamrocker Web HID APIに特有の挙動があるため、Web HID API の都合やその他制約事項から、必要となる UI の要素を導き出してみました。言葉での羅列なのでわかりにくいかもしれません。時間を取って説明をさせてください。

adamrocker, 2020-12-25:
@yoichiro 編集中のキーボードの接続が切れた場合の挙動はどうしましょうか？

・切断により接続・認可済みキーボードが0台になった場合→状態aに（その時のUIも必要？）
・切断により接続・認可済みキーボードが1台になった場合→残ったキーボードを状態cに
・切断により接続・認可済みキーボードが2台以上になった場合→何を基準にして接続キーボードを決めて状態eにしますか？

yoichiro, 2020-12-25:
@adamrocker 接続が切れた際には、編集を継続せずに終了して（まだFlushしていない場合は破棄）、前の状態に戻る、っていうのが基本方針かなと思います。

> ・切断により接続・認可済みキーボードが0台になった場合→状態aに（その時のUIも必要？）

はい。元々1台もキーボードが接続されていなかった場合のUIの対応は必要かと思います。イメージとしては、キーレイアウトは何も表示されず、下部のキー割り当て候補がDisableになっている状態になるかな、と思ってます。

> ・切断により接続・認可済みキーボードが1台になった場合→残ったキーボードを状態cに

はい、残った認可済みキーボードの編集が開始されると良さそうです。

> ・切断により接続・認可済みキーボードが2台以上になった場合→何を基準にして接続キーボードを決めて状態eにしますか？

Remapの起動時に複数の認可済みキーボードがある状態と同じかと思います。つまり、ユーザに「どのキーボードの編集をしますか？」と問い合わせるのが良いかなと思います。

いかがでしょうかー？

adamrocker, 2020-12-25:
良いと思います。
eの状態で状態遷移が11では無い状態（状態遷移12?）で、UIは2を適用するということで理解しました〜

adamrocker, 2020-12-25:
@yoichiro キーボードの接続された状態でキーボードレイアウトの定義が存在しない状態も考慮したほうが良いですかね？

yoichiro, 2020-12-25:
> キーボードの接続された状態でキーボードレイアウトの定義が存在しない状態も考慮したほうが良いですかね？

@adamrocker あー、はい、そうですね。失念してましたすみません。上記にて「キーボードの編集を開始する」というタイミングにて「キーボードレイアウトの定義が存在しない場合にどうするか」となりますね。「キーボードレイアウト定義JSONファイルのアップロード」をすることで編集開始ができるようにしましょう。あ、そうか、そのファイルを持ってなければアップロードできないので・・・アップロード画面をキャンセルできないようにしてしまうくらいしか思いつきません。。。何かアイディアありますでしょうか？

adamrocker, 2020-12-26:
@yoichiro キーボード画面を表示せず「キーボード設定ファイルが見つかりませんでした、設定ファイル（JSON）をアップロードしてください」というメッセージとアップロード用のUIを表示しましょうか？
その際、キーコード側（下半分）はdisabled。

adamrocker, 2020-12-26:
アップロードされたJSONファイルが本当に接続されたキーボードのものかは確認しようがないので、信用するしか無いですね。Vendor ID・Product IDが一致するぐらいは確認しますが…

yoichiro, 2020-12-26:
> キーボード画面を表示せず「キーボード設定ファイルが見つかりませんでした、設定ファイル（JSON）をアップロードしてください」というメッセージとアップロード用のUIを表示しましょうか？
その際、キーコード側（下半分）はdisabled。

@adamrocker はい、そうしましょう！

> アップロードされたJSONファイルが本当に接続されたキーボードのものかは確認しようがないので、信用するしか無いですね。Vendor ID・Product IDが一致するぐらいは確認しますが…

@adamrocker はい、JSONファイルに書かれている Vendor ID と Product ID のチェックのみで大丈夫かと思います。

yoichiro, 2021-01-09:
起動からキーボードの編集開始までの状態遷移:

* 起動時 → 接続開始 → 接続完了 → 定義読み込み開始 → 定義読み込み完了 → オープン開始 → オープン完了

yoichiro, 2021-01-09:
# 起動時

## state

* connectingKeyboard = false
* fetchingKeyboardDefinition = false
* openingKeyboard = false
* keyboardDefinition = null
* keyboard = null

## UI

* KeyboardList = visible
* Keymap = hidden
* in-progress = hidden

# 接続開始

## state

* connectingKeyboard = true
* fetchingKeyboardDefinition = false
* openingKeyboard = false
* keyboardDefinition = null
* keyboard = null

## UI

* KeyboardList = visible
* Keymap = hidden
* in-progress = visible

# 接続完了

## state

* connectingKeyboard = false
* fetchingKeyboardDefinition = false
* openingKeyboard = false
* keyboardDefinition = null
* keyboard = instance(not opened)

## UI

* KeyboardList = visible
* Keymap = hidden
* in-progress = hidden

# 定義読み込み開始

## state

* connectingKeyboard = false
* fetchingKeyboardDefinition = true
* openingKeyboard = false
* keyboardDefinition = null
* keyboard = instance(not opened)

## UI

* KeyboardList = visible
* Keymap = hidden
* in-progress = visible

# 定義読み込み完了

## state

* connectingKeyboard = false
* fetchingKeyboardDefinition = false
* openingKeyboard = false
* keyboardDefinition = instance
* keyboard = instance(not opened)

## UI

* KeyboardList = visible
* Keymap = hidden
* in-progress = hidden

# オープン開始

## state

* connectingKeyboard = false
* fetchingKeyboardDefinition = false
* openingKeyboard = true
* keyboardDefinition = instance
* keyboard = instance(not opened)

## UI

* KeyboardList = visible
* Keymap = hidden
* in-progress = visible

# オープン完了

## state

* connectingKeyboard = false
* fetchingKeyboardDefinition = false
* openingKeyboard = false
* keyboardDefinition = instance
* keyboard = instance(opened)

## UI

* KeyboardList = hidden
* Keymap = visible
* in-progress = hidden

yoichiro, 2021-01-09:
話し合った結果、複数のフラグを使わずに、一つのenumで扱うことに決定。

yoichiro, 2021-01-09:
# 起動時

## state

* setupPhase = keyboardNotSelected
* keyboardDefinition = null
* keyboard = null

## UI

* KeyboardList = visible
* KeyboardDefinitionUploadForm = hidden
* Keymap = hidden
* in-progress = hidden

# 接続開始

## state

* setupPhase = connectingKeyboard
* keyboardDefinition = null
* keyboard = null

## UI

* KeyboardList = visible
* KeyboardDefinitionUploadForm = hidden
* Keymap = hidden
* in-progress = visible

# 接続完了

## state

* setupPhase = connectingKeyboard
* keyboardDefinition = null
* keyboard = instance(not opened)

## UI

* KeyboardList = visible
* KeyboardDefinitionUploadForm = hidden
* Keymap = hidden
* in-progress = hidden

# 定義読み込み開始

## state

* setupPhase = fetchingKeyboardDefinition
* keyboardDefinition = null
* keyboard = instance(not opened)

## UI

* KeyboardList = visible
* KeyboardDefinitionUploadForm = hidden
* Keymap = hidden
* in-progress = visible

# 定義読み込み完了（定義あり）

## state

* setupPhase = fetchingKeyboardDefinition
* keyboardDefinition = instance
* keyboard = instance(not opened)

## UI

* KeyboardList = visible
* KeyboardDefinitionUploadForm = hidden
* Keymap = hidden
* in-progress = hidden

# 定義ファイルアップロード待機

## state

* setupPhase = waitingKeyboardDefinitionUpload
* keyboardDefinition = null
* keyboard = instance(not opened)

## UI

* KeyboardList = hidden
* KeyboardDefinitionUploadForm = visible
* Keymap = hidden
* in-progress = hidden

# 定義ファイルアップロード完了

## state

* setupPhase = waitingKeyboardDefinitionUpload
* keyboardDefinition = instance
* keyboard = instance(not opened)

## UI

* KeyboardList = visible
* KeyboardDefinitionUploadForm = hidden
* Keymap = hidden
* in-progress = hidden

# オープン開始

## state

* setupPhase = openingKeyboard
* keyboardDefinition = instance
* keyboard = instance(not opened)

## UI

* KeyboardList = visible
* KeyboardDefinitionUploadForm = hidden
* Keymap = hidden
* in-progress = visible

# オープン完了

## state

* setupPhase = openingKeyboard
* keyboardDefinition = instance
* keyboard = instance(opened)

## UI

* KeyboardList = hidden
* KeyboardDefinitionUploadForm = hidden
* Keymap = visible
* in-progress = hidden

yoichiro, 2021-01-09:
setupPhase 値によるUIの遷移:

* keyboardNotSelected
  * KeyboardList
* connectiongKeyboard
  * KeyboardList
  * in-progress
* fetchingKeyboardDefinition
  * KeyboardDefinitionUploadForm
  * in-progress
* waitingKeyboardDefinitionUpload
  * KeyboardDefinitionUploadForm
* openingKeyboard
  * KeyboardList
  * in-progress
* openedKeyboard
  * Keymap

closed 2021-01-12 by yoichiro