remap-keys/remap #39: How to register a Keyboard Definition File

yoichiro, 2020-12-26:
This Remap product needs many JSON files which the following information are defined about each keyboard:

* Keyboard Model Information: Vendor ID and Product ID
* The number of keys: Row count and Column count
* The Key Layout: Each key position, size and related row/column numbers

We need to discuss about how to register such JSON file by keyboard designers.

このRemapプロダクトは、各キーボードについて以下の情報が定義された多くのJSONファイルを必要とします。

* キーボードモデル情報: Vendor ID および  Product ID
* キー数: 列数および行数
* キーレイアウト: 各キーの位置、大きさ、そして対応する列/行番号

キーボード作者によってそれらのJSONファイルがどう登録されるのか、議論する必要があります。

yoichiro, 2020-12-27:
初期スコープでは、キーボード定義JSONファイルの登録フォームなどは設けずに、Remap 開発陣にて手動で登録する方式を採用しようと思っています。ただし、バックエンドに登録された情報に基づいて接続されたキーボードを自動認識して編集が開始されるようにはしておきたいと思います。

@adamrocker いかがでしょうか？

adamrocker, 2020-12-27:
@yoichiro 良いと思います。運営側がボトルネックにならないよう、できるだけ早めにキーボード開発者に設定ファイルを自由に登録できるようにしたいですね

yoichiro, 2020-12-27:
@adamrocker そうですね、今回の Remap の開発のモチベーションが「キーボード作者とユーザとのスムーズな連携」なので、初期リリース後速やかに登録フローを整備しましょう。

yoichiro, 2021-01-07:
初期リリースでは、Firebase Firestoreをキーボード定義JSONファイルの格納場所とします。

当初AWS Amplifyを検討しましたが、AWS CognitoがGitHubアカウントでのユーザ認証をサポートしていない（GitHubアカウント認証はOpenID Connectを未サポート）ために、将来的にキーボード開発者の判断（= QMK FirmwareのContributorかどうかの判断）としてGitHubアカウントを利用したいと考えていることから、採用を見送った方が良いと判断しました。

yoichiro, 2021-01-12:
昨日話したことのポイント:

* キーボード定義の登録に対して、承認などの関所を設けないようにする。
* 初期リリース後に、登録フォームを提供する。
* 登録フォームには、GitHubアカウントによるユーザ認証を組み込んで、キーボード定義を登録した人がGitHubでの誰なのかを後で追跡できるようにしておく。
* 登録フォームには、キーボード定義の登録者を後でチェックすることがあること、チェックの結果によってはキーボード定義を削除することもあること、を記載して警告しておく。
* 初期リリース前の限定公開（？）の段階では、登録フォームは提供せず、別の方法でキーボード定義ファイルをキーボード開発者から入手して、手動登録する。

yoichiro, 2021-01-21:
https://docs.google.com/document/d/1zIPqE0A5no4j4ZI3paQaa-KpuN65e7a3DyxXpS5_ekc/edit#

yoichiro, 2021-01-28:
Released at #215

closed 2021-01-28 by yoichiro