remap-keys/remap #175: Support keycap labels for each language

yoichiro, 2021-01-16:
There are an option to change mainly symbols sent by pressing a Shift key which can set by OS. For example, when the "US Keyboard" is set and pressing the "Shift + 2", the "@" symbol is sent. In the other hand, if the "JIS Keyboard" is set and pressing the "Shift + 2", the double quotation symbol is sent.

We guess that Remap users can recognize more easily what is the "2" key if the keycap is shown as "2 and @".

yoichiro, 2021-02-22:
@adamrocker I created the language definition file. It was created automatically by using my script, and the script read the keymap-[lang].h files in the QMK Firmware.

[languages.json.zip](https://github.com/remap-keys/remap/files/6020204/languages.json.zip)

yoichiro, 2021-02-25:
@adamrocker I would like to send you the updated `languages.json` file. In this file, each modifier represents as an array like `"modifiers":["RightShift", "RightCtrl"]`.

[languages.zip](https://github.com/remap-keys/remap/files/6039679/languages.zip)

adamrocker, 2021-02-25:
@yoichiro Thanks! I'm gonna implement them into our app ;)

closed 2021-02-27 by adamrocker