remap-keys/remap #221: Keep the setting of the layout option.

yoichiro, 2021-01-24:
As title. Probably, we may store the setting values into the local storage of the Chrome browser.

adamrocker, 2021-01-25:
[memo for me]
need to check the saved settings are available to the current keyboard definition.

tmor, 2021-03-25:
First of all, thank you for useful service.

I'm using a keyboard with JIS layout.
Currently, every time I open Remap, it reverts to US layout.
I want to save the JIS layout.

My env:
- keyborad: jisplit89
- OS: Win 10 Pro 2004
- Vivaldi: 3.7.2218.45 (Stable channel) (64-bit)

[image]

yoichiro, 2021-05-06:
@tmor This issue covers layout options for keys, does not cover the language option. I will create another issue for the language selection.

PR #450 (2021-05-06): Save/Restore layout option values
Fix #221. This pull request provides the code set to save/restore selected layout option values from/to the keyboard eeprom.
closed 2021-05-06 by yoichiro