remap-keys/remap #294: Application stops if keymap contains undefined keycode

sekigon-gonnoc, 2021-02-16:
- To reproduce
  - Set undefined keycode by using VIA configurator or initial keymap on build
    - "undefined keycode" means keycode is not included in the table defined in remap
    - e.g. Set ANY(0x5D11) to layer 1
  - Connect keyboard to remap, and select layer including undefined keycode.
  - Keymap will not be rendered.
 
- log
```
index.ts:198 strict mode: "items" is 1-tuple, but minItems or maxItems/additionalItems are not specified or different
Y @ index.ts:198
index.ts:198 strict mode: "items" is 2-tuple, but minItems or maxItems/additionalItems are not specified or different
Y @ index.ts:198
index.ts:198 strict mode: "items" is 2-tuple, but minItems or maxItems/additionalItems are not specified or different
Y @ index.ts:198
WebHid.ts:159 1104 0000 0000 0000 0000 0000 0000 0000  ................
0000 0000 0000 0000 0000 0000 0000 0000  ................

WebHid.ts:159 1200 001C 002B 0014 001A 0008 0015 0017  .....+..........
00E0 0004 0016 0007 0009 000A 00E1 001D  ................

WebHid.ts:159 1200 001C 002B 0014 001A 0008 0015 0017  .....+..........
00E0 0004 0016 0007 0009 000A 00E1 001D  ................

WebHid.ts:121 Duplicate response was handled. Ignore.
(anonymous) @ WebHid.ts:121
WebHid.ts:159 1200 1C1C 001B 0006 0019 0005 0000 0000  ................
0000 00E3 5101 002C 002A 0013 0012 000C  ....Q..,.*......

WebHid.ts:159 1200 381C 0018 001C 0034 0033 000F 000E  ..8......4.3....
000D 000B 00E5 0038 0037 0036 0010 0011  .......8.7.6....

WebHid.ts:159 1200 540C 0000 0000 0000 00E6 5102 0028  ..T.........Q..(
0000 0000 0000 0000 0000 0000 0000 0000  ................

WebHid.ts:159 1200 601C 0029 001E 001F 0020 0021 0022  ..`..)..... .!."
00E0 0000 0000 0000 0000 0000 00E1 0000  ................

WebHid.ts:159 1200 7C1C 0000 5D07 5D08 5D11 0000 0000  ..|...].].].....
0000 00E3 0001 002C 002A 0027 0026 0025  .......,.*.'.&.%

WebHid.ts:159 1200 981C 0024 0023 0000 0000 004F 0052  .....$.#.....O.R
0051 0050 0000 0000 0000 0000 0000 0000  .Q.P............

WebHid.ts:159 1200 B40C 0000 0000 0000 00E6 5103 0028  ............Q..(
0000 0000 0000 0000 0000 0000 0000 0000  ................

WebHid.ts:252 SyntaxError: Unexpected token u in JSON at position 0
    at JSON.parse (<anonymous>)
    at e.value (Composition.ts:1300)
    at Function.value (KeycodeList.ts:98)
    at Function.value (KeycodeList.ts:138)
    at WebHid.ts:235
    at Array.forEach (<anonymous>)
    at WebHid.ts:225
    at l (runtime.js:63)
    at Generator._invoke (runtime.js:293)
    at Generator.next (runtime.js:118)
hid.action.ts:347 Object
hid.action.ts:348 layer:1, rowCount:8, colCount: 6, layerCount: 4
hid.action.ts:351 Uncaught (in promise) something wrong in loading kerymaps
WebHid.ts:159 1200 C01C 0029 021E 021F 0220 0221 0222  .....)..... .!."
00E0 0000 0000 0000 0000 0000 00E1 0000  ................

WebHid.ts:159 1200 DC1C 0000 0000 0000 0000 0000 0000  ................
0000 00E3 5103 002C 002A 0227 0226 0225  ....Q..,.*.'.&.%

WebHid.ts:159 1200 F81C 0224 0223 0035 0231 0230 022F  .....$.#.5.1.0./
002E 002D 0235 0031 0030 002F 022E 022D  ...-.5.1.0./...-

WebHid.ts:159 1201 140C 0000 0000 0000 00E6 0001 0028  ...............(
0000 0000 0000 0000 0000 0000 0000 0000  ................

WebHid.ts:159 1201 201C 5C00 0000 0000 0000 0000 0000  .. .\...........
5CC2 5CC5 5CC7 5CC9 0000 0000 5CC3 5CC6  \.\.\.\.....\.\.

WebHid.ts:159 1201 3C1C 5CC8 5CCA 0000 0000 0000 0000  ..<.\.\.........
0000 00E3 0001 002C 0000 0000 0000 0000  .......,........

WebHid.ts:159 1201 581C 0000 0000 0000 0000 0000 0000  ..X.............
0000 0000 0000 0000 0000 0000 0000 0000  ................

WebHid.ts:159 1201 740C 0000 0000 0000 00E6 0001 0028  ..t............(
0000 0000 0000 0000 0000 0000 0000 0000  ................

react-dom.production.min.js:216 TypeError: Cannot use 'in' operator to search for '0,3' in undefined
    at Keymap.tsx:350
    at Array.forEach (<anonymous>)
    at qr (Keymap.tsx:348)
    at oa (react-dom.production.min.js:157)
    at Ba (react-dom.production.min.js:180)
    at Vc (react-dom.production.min.js:269)
    at Ss (react-dom.production.min.js:250)
    at zs (react-dom.production.min.js:250)
    at Es (react-dom.production.min.js:250)
    at bs (react-dom.production.min.js:243)
cc @ react-dom.production.min.js:216
```

yoichiro, 2021-02-16:
@sekigon-gonnoc Thank you for the report! I could reproduce this issue on my PC, and we already have found the cause. I think that I might be able to fix this issue within a few days.

closed 2021-02-16 by yoichiro

yoichiro, 2021-02-16:
@sekigon-gonnoc The fixed version 0.2.1 has been released. Could you confirm whether this issue is fixed or not with your environment? Thanks!

PR #295 (2021-02-16): Fix the issue #294
FIx #294. The cause is that there is a keycode within the loose keycode range and the keycode is not defined the keyinfo list.