remap-keys/remap #301: Can't open the "torn" keyboard

yoichiro, 2021-02-17:
Reported by richardt:

> I tried it with my keyboard, but it fails to work after I have uploaded the json file (I just see the remap logo).
> The json file works with via, this is the file I tried https://github.com/the-via/keyboards/blob/master/src/torn/torn.json.

yoichiro, 2021-02-17:
I tried to reproduce this issue, but I noticed that I don't have atmega328p MCU, therefore I can't test that...

yoichiro, 2021-02-17:
We need to get more information to fix this issue. For example, the debug log output on the Chrome DevTools Console is very valuable...

rtitmuss, 2021-02-17:
Here is the debug log:

[remap-keys.app-1613598773392.log](https://github.com/remap-keys/remap/files/5998852/remap-keys.app-1613598773392.log)

Thanks!

yoichiro, 2021-02-17:
@rtitmuss  Thank you for submitting the file. When I confirmed the content, I found the message: "Keyboard definition not found: 29300:1:Torn". This is an expected message, because the Remap backend server does not know the keyboard.

So, after importing your JSON file on the Remap, could you find any error messages on the "Console Tab" of the Chrome DevTools?

[image]

rtitmuss, 2021-02-18:
I used the "Save As..." option in the console. The additional errors are in the log file. Here is a screenshot:

[image]

yoichiro, 2021-02-19:
@rtitmuss Hmm... **after importing your JSON file on the Remap**, could you find any error messages on the DevTools console?

rtitmuss, 2021-02-20:
I made a screen recording with the debug console:

https://user-images.githubusercontent.com/890714/108588055-7fe51280-7357-11eb-81de-0e832a21c1fe.mov

yoichiro, 2021-02-20:
@rtitmuss Thank you for the movie! I can understand the situation. Well, Remap sent the request to your keyboard, but it seems that your keyboard did not send any responses to the Remap. I guess that the firmware of your keyboard does not have the VIA support. Could you confirm whether the firmware has `VIA_ENABLE = yes` or not? And, can you connect your keyboard to VIA app and can customize the keymap?

rtitmuss, 2021-02-20:
Yes, I designed the keyboard, added support for the keyboard in QMK with VIA support, and pushed the keyboard definition to the VIA repo. I tested everything still worked in VIA after trying remap-keys. So I'm 100% sure it's VIA enabled :)

I see you enabled local builds, and have been doing a little debugging myself too. So far I've traced the problem to WebHid fetchLayerCount. I might have time to look more over the weekend, so let me know if you have anything you'd specifically like me to check.

yoichiro, 2021-02-20:
@rtitmuss We improve the Remap to be able to work locally. You can build and launch the Remap on your local environment after setting the environment variable `REACT_APP_WEBHIDAPI_TOKEN="YOUR_WEBHID_ORIGIN_TRIALS_TOKEN"`. Well, I would like to know whether the request sent from Remap has been reached to the QMK Firmware or not. I guess that you can confirm that with QMK Tools for seeing the debug logs. If possible, could you do that?

rtitmuss, 2021-02-27:
I had a chance to look at this today.

The issue is caused by the vusb protocol in qmk (used by atmega328p mcu keyboards). See https://github.com/qmk/qmk_firmware/blob/master/tmk_core/protocol/vusb/vusb.c#L160. It only processes raw hid messages after RAW_BUFFER_SIZE (32) bytes have been received.

I have applied this patch locally and remap works with the Torn keyboard:

diff --git a/src/services/hid/Commands.ts b/src/services/hid/Commands.ts
index 6fc08b5..f11189d 100644
--- a/src/services/hid/Commands.ts
+++ b/src/services/hid/Commands.ts
@@ -40,8 +40,10 @@ export abstract class AbstractCommand<
   async sendReport(device: any): Promise<void> {
     try {
       const outputReport = this.createReport();
-      outputUint8Array('Send data', outputReport);
-      await device.sendReport(AbstractCommand.OUTPUT_REPORT_ID, outputReport);
+      const paddedOutputReport = new Uint8Array(32);
+      paddedOutputReport.set(outputReport);
+      outputUint8Array('Send data', paddedOutputReport);
+      await device.sendReport(AbstractCommand.OUTPUT_REPORT_ID, paddedOutputReport);
     } catch (error) {
       await this.getResponseHandler()({
         success: false,

Please let me know if you'd like me to submit a pull request, or if you will look at fixing this another way.

Thanks!

yoichiro, 2021-02-27:
@rtitmuss Thank you for the investigating! I don't know that it is necessary to send enough bytes (the size is RAW_BUFFER_SIZE) for the atmega328p. At least, we have tested with only atmega32u4.

We need to test your code on atmega32u4. So, could you create a pull request with your code? If it is difficult, of course I can create it instead of you.

rtitmuss, 2021-02-28:
https://github.com/remap-keys/remap/pull/341

yoichiro, 2021-03-01:
@rtitmuss Thank you for creating the pull request! I tested your code on my environment including the atmega32u4 keyboard designed by me, and as the result your code worked normally.

I merged your pull request into the main branch, and it was deployed as the build number 8. Could you test the deployed version 8 on your keyboard?

rtitmuss, 2021-03-01:
Confirmed this works. Thanks.

yoichiro, 2021-03-01:
@rtitmuss Thank you for the confirmation! I close this issue.

closed 2021-03-01 by yoichiro