remap-keys/remap #303: hotdox layout not apearing complete

IsmaelAlvarez, 2021-02-18:
[image]

on each side there are keys not apearing, they are the second last keys ond thumb cluster, im adding via picture for reference

[image]

this definitions used 

https://github.com/the-via/keyboards/tree/master/src/hotdox

yoichiro, 2021-02-18:
Probably, [this JSON file](https://github.com/the-via/keyboards/blob/703e53b6f9d89cf0f3667da2a65afd06e1b2376f/src/hotdox/hotdox.json) is applied. I thought that there are one or more layout option, but nothing...

@adamrocker Could you confirm this issue?

adamrocker, 2021-02-18:
Sure. I assigned this issue to me ;)

adamrocker, 2021-02-18:
@IsmaelAlvarez Thank you for your report. I checked key rendering algo and fixed it.
Fixed version of Remap will be deploy soon.
Please reload Remap web app and make sure it works.

closed 2021-02-18 by adamrocker