remap-keys/remap #347: [Feature Request] Test Matrix

yoichiro, 2021-03-02:
This proposal was provided by @rtitmuss. I quote his suggestion:

> I like the "test matrix" feature in via, I think it's useful when people are building the keyboard to check it is working ok. Is that something you plan to add to remap? I might try to find the time to submit that as a PR if your interested?

The screenshot of the text matrix feature on VIA client app is:

[image]

yoichiro, 2021-03-02:
Well, current Remap doesn't have any key tester features yet. At least, I'm interested in the test matrix feature, but, I think that we should provide the key tester feature with displaying full keys layout before providing the test matrix feature... especially, for users who want to test the customized keymapping...

rtitmuss, 2021-03-02:
I have been thinking about why I used the key tester in VIA, and I think it was for two different use-cases:

1. To make sure I have mapped all the relevant keys. The full key layout can be used to test this, but this seems inefficient. Instead this use-case might be better met by showing which keys are mapped in the main screen for example by changing the background colour of the keys here:

[image]

2. To test that the keyboard hardware is working, especially after building a new keyboard.

If these are the only use-cases that need to be supported, they could be implemented as extensions to the existing remap window.

I am interested to know if there are other use-cases where testing with the full key layout makes sense?

yoichiro, 2021-03-02:
@rtitmuss In most cases, I think the motivation to use testing with the full key layout is 2. Also, I guess that people who don't know VIA is using the "TEST KEYBOARD" page on the QMK Configulator for the same purpose: https://config.qmk.fm/#/test

yoichiro, 2021-03-02:
@rtitmuss BTW, I recognize that we cannot get the press/release events of key codes which are consumed in the firmware only (ex. MO(1) and etc.), because they are not sent to OS. To implement the test matrix feature like provided from VIA client app, Remap needs to get such events. Currently, I don't have any idea how to handle their event via WebHID API.

rtitmuss, 2021-03-03:
@yoichiro it is possible to scan the keyboard matrix with the via protocol. See  https://github.com/qmk/qmk_firmware/blob/master/quantum/via.c#L229

yoichiro, 2021-03-03:
@rtitmuss Oh, I don't know the behavior of the feature to scan each status of switches, thank you. I could understand that we can implement the test matrix feature on Remap (probably, we need to use a polling technique to get each switch status continuously, right?).

And, sorry, I want to ask you about the use cases. How steps is the use case when users use the text matrix feature? And, how steps is the use case when keyboard developers use the text matrix feature?

yoichiro, 2021-03-07:
@rtitmuss I discussed the Text Matrix feature with @adamrocker, and we decided that we start designing and implementing it. Currently, we're focusing users who bought a self-made keyboard kit as the mission of Remap. And, we think that the Text Matrix feature is suitable for the mission.

Before starting the implementation, we need to share the same understanding with us for the UI design of the Text Matrix. Do you already have the idea? If yes, could you share it?

rtitmuss, 2021-03-20:
(Sorry for the slow reply)

I see the key tester in via (& qmk configurator) as providing two use-cases, which I personally think would be better done differently. I think it should be possible to avoid tabs, and have everything in the existing interface.

1. Testing the hardware work (this feature request).

I'd suggest to add a toggle near the flash button, which enables the keyboard scanning. The keys would highlight as you press them, similar to via. For example:

[image]

2. Understanding you've mapped all the keys.

The other reason I've used the test keyboard is to understand I've mapped all the keys. But I shouldn't really need to test the keyboard to do that!? This is maybe another feature request.

Here I would highlight the keys that are mapped in the bottom section of the screen (shown in blue below). It would also be good as you mouse over the keys in the bottom section to highlight the mapped keys on the keyboard (shown in red).

[image]

Optionally you could change the bottom section to have an iso/ansi (or other) layout for configuration, instead of a grid.

I think these suggestions would make remap simpler to use than via / qmk configurator, but equally as useful.

yoichiro, 2021-03-22:
@rtitmuss Thank you for your idea!

When we heard about these features from some users, most users had an opinion which the Test Matrix feature is valuable. Therefore, we decided that we start the design and implementation of the Text Matrix feature. In the other hand, we decided that the key mapping test feature is not included in this scope.

Current Remap has already many features. To improve the UX, we have a plan to re-design the toolbar to divide icons for suitable categories. The current idea is... we add a new icon "..." on the bottom of the toolbar and put the sub category menu of its icon which has some menus to invoke features for during building a keyboard and for developers. We think that the Text Matrix menu should be included in the sub category menu.

We're creating the mock up of the UI. Could you wait it? We think that we need to discuss and agree the UI design with you before starting the implementation.

rtitmuss, 2021-03-22:
Yes, no hurry, happy to wait.

yoichiro, 2021-03-22:
@rtitmuss We have just released the sub menu UI:

[image]

Probably, the menu to invoke the Text Matrix feature should be put into the sub menu.

yoichiro, 2021-03-30:
@rtitmuss Sorry for late reply. @adamrocker designed and implemented the UI for the Test Matrix. You can get the implementation code set from the `feat347-test-matrix` branch.

https://github.com/remap-keys/remap/tree/feat-347-test-matrix

Test Matrix UI can be invoked from the following menu item:

[image]

When clicking the menu item, the Test Matrix UI is shown as like the following:

[image]

The code to show the Test Matrix UI is the following:

https://github.com/remap-keys/remap/blob/feat-347-test-matrix/src/components/configure/keymapToolbar/KeymapToolbar.tsx#L173-L176

When tapping a physical key, you can call the following code to change a color on the top of the target key cap with the row and col position string as the argument:

`_dispatch(AppActions.addTestedMatrix('0,3'));`

https://github.com/remap-keys/remap/blob/2b53411e0290e567e6014b8b0fe39da87c6f52c0/src/components/configure/keymapToolbar/KeymapToolbar.container.ts#L34-L38

Probably, you will write a code to call the command to fetch a key matrix status continuously with `setTimeout` or other such mechanism in a UI component class and to change the color of the target key cap with the code above, I guess.

Could you confirm the code set of the branch? Also, could you implement a new command to fetch the key matrix status and add a new code to reflect the status for each key cap to the `feat-347-test-matrix` branch? 

Thank you for your contribution!

rtitmuss, 2021-03-30:
Hey, so I've created a draft PR here https://github.com/remap-keys/remap/pull/431.

There are a couple of issues that still need to be resolved:
1. Keyboard navigation should be disabled while in test mode, as you can still navigate the ux while testing (for example with keys mapped to tab or enter). I'm not sure how to do this in the browser?
2. Via includes a visual treatment for pressed keys, and released keys. I think both these visual states are needed in case of stuck keys.

Take a look, and let me know what you think!

yoichiro, 2021-03-31:
@rtitmuss Thank you for creating the pull request! I will check it within a few days.

> Keyboard navigation should be disabled while in test mode, as you can still navigate the ux while testing (for example with keys mapped to tab or enter). I'm not sure how to do this in the browser?

Basically, during displaying the Test Matrix mode, it is necessary to handle most key events and destroy them to avoid some behaviors of the browser. That is, I guess that we need to add an event handler to the parent element and handle/destroy them.

> Via includes a visual treatment for pressed keys, and released keys. I think both these visual states are needed in case of stuck keys.

This is a good suggestion. Yes, the current code set doesn't have an interface to clear the highlighted. Therefore, we need to improve the current interface about that.

@adamrocker Could you improve the interface to turn on/off the highlighting of each key top?

yoichiro, 2021-04-01:
@rtitmuss Thank you for the contribution. We have merged your pull request, but actually we reverted the merge. Because, we confirmed that your name was not listed in the contributors. To be listed as a contributor on the GitHub repository, it is necessary to include your commit in the default branch (it is the main branch in the Remap repository).

We want to list your name as the contributors. To do that, could you re-create a pull request? The target branch of the pull request must be to the main branch. Before a few hours, we have already merged the base branch. Currently, you can create a pull request to the main branch.

The sub menu item to invoke the Test Matrix feature is currently hidden. Could you activate the following comment lines in your pull request?

https://github.com/remap-keys/remap/pull/435/commits/fb507df206badd63ede6cb0bd00921a5fa0d3bf0

I'm sorry about duplicate efforts...

rtitmuss, 2021-04-02:
No problem, I've pushed this again here https://github.com/remap-keys/remap/pull/436. Thanks.

yoichiro, 2021-04-02:
@rtitmuss Thank you for the re-creating the pull request! I merged the pull request, then this feature will be released soon.

closed 2021-04-02 by yoichiro

yoichiro, 2021-04-02:
@rtitmuss We found that your name is not listed in the contributors after merging your pull request. When we investigated this problem, we found the reason. That is, it seems that the email string set on your local environment is not matched to the registered it to your GitHub account. We wrote your name on the release note. Thank you for the contribution. :)

rtitmuss, 2021-04-06:
Ah, I guess I need to remember to update that when submitting contributions. Thanks.

PR #433 (2021-04-01): Feat 347 Test Matrix Feature
Fix #347