remap-keys/remap #354: 2U Vertical Keys Split issue

Xelus22, 2021-03-03:
Hi Remap team,

I have an issue with my AEBoards EXT65. The vertical 2U key is misaligned when split into 2x 1U keys.

[image]
[image]
[image]

adamrocker, 2021-03-03:
@Xelus22 Thank you for your report! Is there this keyboard definition file I can access?

Xelus22, 2021-03-03:
Yep the file is here:
https://github.com/the-via/keyboards/blob/master/src/aeboards/ext65/aeboards_ext65.json
Thanks

adamrocker, 2021-03-03:
Thank you for sharing the file. I can understand the situation. I'm gonna fix it :)

adamrocker, 2021-03-04:
@Xelus22 I fixed this bug. Could you please check the behavior with your environment?

closed 2021-03-04 by adamrocker

Xelus22, 2021-03-04:
Yep works perfectly now thank you!
[image]