remap-keys/remap #355: Keycode values for mouse keys are changed

yoichiro, 2021-03-03:
Recently, key code values for the mouse keys are changed in the QMK Firmware code:
https://github.com/qmk/qmk_firmware/pull/10807

Currently, the mouse buttons 1 to 5 are defined in the Remap. However, in the develop branch of QMK Firmware, the new mouse buttons 6 to 8 are added. Therefore, already-defined key codes of other mouse keys will be shifted.

As the result, we will need to support the situation that there are two keyboards. One has the old key code set for the mouse keys. Another one has the new key code set for them. Probably, Remap will have the switch UI to use either of old/new keycode set for mouse keys.

See: https://discordapp.com/channels/802306358938435605/816488260250370080/816494996676411422

yoichiro, 2021-03-03:
The develop branch including the changes has already been merged into the master branch on 27th of this February. That is, we should recognize that such keyboards including the changes already exist...

maz0r, 2021-03-15:
note the following https://github.com/qmk/qmk_firmware/pull/12130 was merged into QMK, reversing the offset key's, but only for firmware built since the fix was committed OR that was manually patched by the user/creator.

yoichiro, 2021-03-15:
@maz0r Thank you for the information. Hmm... Currently, the huge problem is that we don't have a way to get the QMK Firmware version of connected devices. Therefore, Remap can't adjust the correct key code set automatically...

yoichiro, 2021-05-07:
Basically, the current Remap is following the latest QMK Firmware which some key codes have already been restored.

closed 2021-05-07 by yoichiro