remap-keys/remap #356: [Feature Request] RGBLIGHT mode changer

Xelus22, 2021-03-03:
Currently VIA allows support of changing RGBLIGHT as a GUI.

[image]
[image]

yoichiro, 2021-03-03:
@Xelus22 The lighting feature has already been provided on Remap. If you click the gear icon,

[image]

you can see the control dialog to set each lighting configuration. Could you confirm this feature on your environment and your keyboard?

[image]

Xelus22, 2021-03-03:
Oh, my keyboard doesnt have RGB effects. Sorry about that. Didn't know it was already implemented.
Would it be better to have another tab instead of in the "settings" ?

yoichiro, 2021-03-03:
> Would it be better to have another tab instead of in the "settings" ?

@Xelus22 Yeah, we may need to improve the UI.

@adamrocker Let's discuss this issue as well during the UI improvement including removing the menu.

yoichiro, 2021-03-03:
@Xelus22 I close this issue. We intend to continue the discussion about the UI improvement on #357. Thanks!

closed 2021-03-03 by yoichiro