remap-keys/remap #357: Improve the UI (mainly the menu)

yoichiro, 2021-03-03:
Currently, there is a menu UI component on the top of the right. The menu has one menu item to export a cheat sheet.

[image]

If we add a new menu to provide some menu items about user's preferences at the right on the header after releasing a login feature, two menus will be put on very near place. This is not good.

Also, if we put icons to invoke each function as like a general toolbar (that is, not use the menu component), users will be able to invoke them more easily than now.

yoichiro, 2021-03-03:
@adamrocker In the discussion #356, the issues reporter could not notice the existence of the lighting configuration feature. Well, I guess that we should put an icon to open the lighting feature dialog on the icons bar discussing on this issue.

adamrocker, 2021-03-04:
@yoichiro Good idea! I agree with you.

adamrocker, 2021-03-04:
I've just added the issue which puts the lighting menu on the feature menu.
https://github.com/remap-keys/remap/issues/359

closed 2021-03-04 by adamrocker