remap-keys/remap #390: Parser incorrectly handles valid VIA definition

rys, 2021-03-11:
In `KeyboardModel`'s `parseKeyMap()`, there's an incorrect assumption that after parsing a `KeyOp`, the next item in the definition will always be a string label. Unfortunately that's an incorrect assumption, since VIA allows successive `KeyOps` to follow each other.

[KeyboardModel.ts#L226](https://github.com/remap-keys/remap/blob/daa91e0f3f79b4bae88ba734679adcd39d39793d/src/models/KeyboardModel.ts#L226) is where the error occurs, causing an uncaught exception in the`KeymapItem` constructor when it tries to call `split()` on `label`, which isn't a string.

If you want an example VIA definition that doesn't load properly, try the [Think6.5](https://raw.githubusercontent.com/the-via/keyboards/master/src/graystudio/think65/think65.json). The issue is in the first row, where `{ "c": "#cccccc" }, { "x": 0.5 }` causes the problem. Sorry I'm not confident enough with TypeScript to propose a fix!

adamrocker, 2021-03-12:
@rys Thank you for your report. I could understand the specification. I'll fix it!

adamrocker, 2021-03-12:
@rys Hi. I fixed it and deploy the app soon.
Could you please check it?
Thank you.

closed 2021-03-12 by adamrocker

rys, 2021-03-12:
Works great now, thanks!