remap-keys/remap #400: Large Layout with small screen

AndyChiu, 2021-03-15:
Hi, I found a problem, when my keyboard is too large, and screen too small, the web page cannot accommodate the layout, the web page screen cannot be dragged like the KLE web side. The tool buttons can only be displayed when the web  page is reduced to 50%. However, the screen cannot be seen clearly when use 50%. Is there a way to improve this problem?  Thank you!

[image]
can't move screen to see all layout and choice tool button

[image]
resize web page is ok but can't seen clearly

[image]
KLE can move and see all layout and function button

yoichiro, 2021-03-15:
@AndyChiu To reproduce this issue, could you share us the JSON file and the hex file?

AndyChiu, 2021-03-15:
Yes! Tks!
[Universal-Century(willow64+extension-to-108)-REMAP.zip](https://github.com/remap-keys/remap/files/6139231/Universal-Century.willow64%2Bextension-to-108.-REMAP.zip)

[UC_default.zip](https://github.com/remap-keys/remap/files/6139246/UC_default.zip)

AndyChiu, 2021-03-15:
the screen size is 1024*768

yoichiro, 2021-03-15:
@AndyChiu Thank you for the sharing.

adamrocker, 2021-03-16:
@AndyChiu Hi Andy,
Thank you for using Remap and report the bug. I fixed it :) Please check it on build:43. Thanks.

AndyChiu, 2021-03-17:
Hi, now i try build:44
the have vertical scoll bar, but don't have horizontal scroll bar, i can use arrow key to control horizontal scroll, but if have horizontal scroll bar is better!

[image]

yoichiro, 2021-03-17:
@adamrocker Hmm... Could you confirm whether this issue was fixed or not in the latest revision?

adamrocker, 2021-03-17:
@AndyChiu Thank you for checking that. At this moment, please use left/right arrow key to scroll right and left.
I can move for side direction using mouse scroll and those keys.

I'm considering to fix to show the side scroll bar.
Sorry for the inconvenience this may have caused.

adamrocker, 2021-03-17:
@AndyChiu Again, I fixed it. Please check it when you have available time :)

closed 2021-03-17 by adamrocker

AndyChiu, 2021-03-17:
@adamrocker Thank you! so fast so nice!  👍