remap-keys/remap #448: 7skb keyboard layout is not drawn correctly

yoichiro, 2021-04-28:
[image]

When selecting the `Choc` layout option, some keys were not drawn correctly.

yoichiro, 2021-04-28:
@adamrocker Could you confirm this issue? Thanks.

yoichiro, 2021-04-28:
I captured the window of VIA, when I tested this behavior on VIA client app.

[image]

[image]

adamrocker, 2021-04-28:
I investigated the JSON and found that it has the wrong format.
The "9,0" key does not have an "Option Choice" label, on the other hand, the VIA's JSON has it.
The size of choice keys: "4,4", "9,0", are also wrong.

The layout of uploaded JSON is as below:
[image]

The VIA's JSON is as below:
[image]

yoichiro, 2021-05-07:
I close this issue because this issue caused by the specific keyboard definition.

closed 2021-05-07 by yoichiro