remap-keys/remap #449: "MUST have .json extention" error though it's json

miyamonz, 2021-04-29:
## What happened

I'm using moonlander, which isn't supported by Remap.
so I prepared its JSON file.
but it failed.
[image]

## why
I surveyed this and found the reason.
the File.type is not `application/json` but `text/plain` on my environment. Windows 10, chrome.

this is a very small reproduction.
https://codesandbox.io/s/dank-fog-k6w6v
I tested on this.
when it's windows, it's text/plain
when it's Mac, it's application/json with a same json file.

I don't check the detail, but probably the File.type isn't correct.

ref: https://developer.mozilla.org/en-US/docs/Web/API/File/type

## suggestion
I don't know why such a difference happens, and understand it's not Remap's bug.
anyway, modifying the below will fix this problem.

https://github.com/remap-keys/remap/blob/693c38c2883991b974f10c1e32d9f2cde4deca75/src/services/storage/Validator.ts#L25-L27

like this:
```js
function isJSON(sJSON){
    try {
        JSON.parse(sJSON);
        return true;
    } catch (e) {
        return false;
    }
}
```

if it's OK, I will send pull req.

yoichiro, 2021-05-06:
@miyamonz Thank you for this report and the investigation. I agree the modification you suggested. Could you create a pull request?

yoichiro, 2021-05-26:
@miyamonz I have just merged the pull request #452. Could you test the latest behavior?

yoichiro, 2021-05-29:
I close this issue.

closed 2021-05-29 by yoichiro

miyamonz, 2021-06-01:
yes! it works. thank you!