remap-keys/remap #455: Attack25 rev3 layout broken

monksoffunk, 2021-06-02:
[image]
Layout broken when layout option 'Encoder' is off.

https://discord.com/channels/802306358938435605/808957769415786536/849473200995696661

yoichiro, 2021-06-05:
@adamrocker It seems that there is something issue about rendering the key layout with a layout option. Could you investigate and try to fix this issue?

yoichiro, 2021-06-14:
@monksoffunk We're investigating this issue. But, we guess that it will take a long time a bit to fix this issue completely. As the alternate way, this issue does not occur when the rotary encoder is shown as default. If you want to avoid this issue soon, could you modify the JSON file to change the default behavior to show the rotary encoders?

closed 2021-06-14 by adamrocker