remap-keys/remap #456: PDF output failed

yoichiro, 2021-06-05:
When outputting a key mapping PDF file for the specific keyboard, the following error occurrs:

```
KeymapPdfGenerator.ts:283 Uncaught (in promise) TypeError: Cannot read property 'label' of undefined
    at e.value (KeymapPdfGenerator.ts:283)
    at KeymapPdfGenerator.ts:170
    at Array.forEach (<anonymous>)
    at e.value (KeymapPdfGenerator.ts:166)
    at e.value (KeymapPdfGenerator.ts:132)
    at e.<anonymous> (KeymapPdfGenerator.ts:100)
    at l (runtime.js:63)
    at Generator._invoke (runtime.js:293)
    at Generator.next (runtime.js:118)
    at r (asyncToGenerator.js:3)
```

The target keyboard is not registered into the Remap yet.

Hansyo, 2021-06-05:
アップロードしたjsonファイルです
https://gist.github.com/Hansyo/91b2c33fa7c335dbb5e35db1bb62299d

アップロードした当時のコンソール出力です
https://gist.github.com/Hansyo/bb1f41cda0322f1fc2445902f81836f6

yoichiro, 2021-06-05:
@Hansyo Thank you for the reporting this issue and sharing them. We will investigate and try to fix about this issue.

Hansyo, 2021-06-10:
@yoichiro
私のアップロードしたjsonファイルが間違っていました。
jsonを正しいファイルに訂正したところ、きちんとPDFを出力することができました。

runner3680は、作成者によってkeymapが変わるキーボードです。
私の場合(row x line)の順で、 正しくは**(5 x 7) が両手分** のところ、アップロードしたjsonファイルでは **(5 x 8) を両手分** で記述していました。

正しいjsonファイルは、以下のgistにあります。
https://gist.github.com/Hansyo/eeaa89b9540d303a5e72dd734a77de7a

yoichiro, 2021-06-10:
@Hansyo Thank you for the feedback. I understand that the reason is that the col or row number of the JSON file does not match them of the firmware.

However, I think that there is other issue. when you tried to output a PDF in the wrong situation, the Remap did not show any error message. I think that the Remap should show an error message. I intend to fix this issue. So, I will close this issue after fixing that.

closed 2021-06-15 by adamrocker