remap-keys/remap #460: [Documentation] json file format documentation is not easy to locate

jaygreco, 2021-06-21:
We're exploring adding remap support for our nullbits products as a VIA alternative. There is currently no documentation that is easily discoverable (<5min) that details the process of adding a new keyboard or porting from a VIA json file. This documentation will help encourage adoption.

yoichiro, 2021-06-28:
@jaygreco Sorry for this late reply. Basically, you can send a review request from the following URL with the VIA JSON file without any modification: https://remap-keys.app/keyboards But I intend to write a document to describe how to register a keyboard to Remap soon. Please give me a few days.

jaygreco, 2021-06-29:
Not a problem at all! I was able to find the spec and add a keyboard. This would still be helpful for those who are looking to port support from VIA. 

Thanks for the response!

closed 2022-01-02 by jaygreco