remap-keys/remap #480: Working with Report ID

yswallow, 2021-07-03:
I'm making keyboard firmware.

I want to use Remap via BLE. I implemented BLE raw HID and it works with my application.

BLE HID report must divided by Report ID because it has no interface feature.

I found Remap is using Report ID 0x00 constantly. 
https://github.com/remap-keys/remap/blob/004e572d9f2092c7d41418196ad09a45600b7995/src/services/hid/Commands.ts#L16

Can you modify it to using report id which points vendor-defined report ?

yoichiro, 2021-07-09:
@yswallow Hmm... I didn't notice that because I tested with keyboards using ProMicro with ATmega32u4 only. That is, when communicating with your keyboard via BLE raw HID, we need to use the value except 0x0000, right? Well, what concrete number is it? And, is there some way Remap can detect the value dynamically?

yswallow, 2021-07-10:
> we need to use the value except 0x0000, right?

Right. 

I'm thinking of using report id 8.

And we can find report id in `device[0].collections[0].outputReports[0].reportId` .

```javascript
filters = [{}];
device = await navigator.hid.requestDevice({filters});
// select USB RAW HID device
device[0].collections[0].outputReports[0].reportId
// =>0

device = await navigator.hid.requestDevice({filters});
// select my BLE RAW HID device
device[0].collections[0].outputReports[0].reportId
// =>8
```

yoichiro, 2021-07-10:
@yswallow Thank you for the information. I intend to modify the code to use the report ID which is returned after the keyboard is connected.

yswallow, 2021-07-11:
@yoichiro Thank you.

yoichiro, 2021-07-12:
@yswallow I created the pull request #496 to use an output report ID returned from a connected keyboard. If the code is correct, your firmware should work correctly with the Remap. Could you test the code set of the #496 on your local environment?

yswallow, 2021-07-12:
@yoichiro It works fine!! Thank you!!

[image]

However, it is not a remap's problem, device name displayed as too many TOFUs.
This is probably Windows' problem in `HidD_GetProductString`.

I may need to commit Chromium....

yoichiro, 2021-07-12:
@yswallow Thank you for the confirmation. Ok, I will release the code to fix this issue. But, when something wrong occurs after releasing it, I will revert the code, because I can't test all keyboards and it might be possible some error occurs. Probably, I guess that the code works...

closed 2021-07-14 by yoichiro

yoichiro, 2021-07-16:
@yswallow I don't know the reason why the keyboard name became tofus, but I have just released a new revision of Remap to fix that the header UI was broken by the long keyboard name.

PR #496 (2021-07-14): Introduce w3c-web-hid type def and use the output report ID.
This pull request fixes #480 

Currently the Remap is using the fixed value `0x00` as an output report ID. However, there are keyboards which are using other value as the output report ID. As mentioned in #480, we should use the output report ID which is included in the collections returned connecting a keyboard.

Also, to reduce bugs, I introduce the w3c-web-hid type definition.