remap-keys/remap #489: Warning of using development build of firebase JS SDK

LSChyi, 2021-07-06:
Is there any special reason to use the development build of the firebase JS SDK? I saw the warning message in the console:

```
It looks like you're using the development build of the Firebase JS SDK.
When deploying Firebase apps to production, it is advisable to only import
the individual SDK components you intend to use.

For the module builds, these are available in the following manner
(replace <PACKAGE> with the name of a component - i.e. auth, database, etc):

CommonJS Modules:
const firebase = require('firebase/app');
require('firebase/<PACKAGE>');

ES Modules:
import firebase from 'firebase/app';
import 'firebase/<PACKAGE>';

Typescript:
import firebase from 'firebase/app';
import 'firebase/<PACKAGE>';
```

which is caused by the package `src/services/auth/Auth.ts`, which imports the firebase by `import firebase from 'firebase';`, while the `src/services/provider/Firebase.ts` is using the production build with import the firebase by `import firebase from 'firebase/app';`?

yoichiro, 2021-07-09:
@LSChyi Thank you for the report. Well, could you create a pull request to fix this issue?

LSChyi, 2021-07-09:
Sure, I will do it today.

yoichiro, 2021-07-09:
@LSChyi Great! :)

yoichiro, 2021-07-10:
@LSChyi I have merged your pull request. Thank you for your contribution! I close this issue.

closed 2021-07-10 by yoichiro

LSChyi, 2021-07-11:
Great thanks!