remap-keys/remap #493: レイアウト変更時にツールが選択しづらくなる

mokesura, 2021-07-09:
お世話になっております。
自設計のキーボードでRemapを試しています。

jsonファイルでレイアウトを2種類（左右反転）作成したのですが、Remapでレイアウトを「Left Hand」から「Right Hand」に切り替えると、画像の赤丸を付けたところが選択しづらくなってしまいます。

（まったく選択できないわけではなく、ごく小さい範囲でしかマウスカーソルの形が変わらず、それ以外の場所ではクリックしても反応がありません。また、その状態でもキーマップの変更、書き込みは行えます）

[image]

他の、Remapに登録されていて、レイアウトオプションがあるキーボード（te96）のファームウェアを書き込んで試してみたところ、そういう状態にはなりませんでした。

原因がどこにあるか、ご確認いただけますでしょうか。

---
事象が発生しているファイルは以下です。

- hexファイル  
[cc_proto_default.zip](https://github.com/remap-keys/remap/files/6790618/cc_proto_default.zip)

- jsonファイル  
https://github.com/mokesura/cc_proto/blob/da2882e17f20853f98954cefcdbcf79a92b22269/via_json/cc-proto_via.json

また、レイアウトを変更したときの、Chromeでの「Console」タブのメッセージはこちらです。
- [console.txt](https://github.com/remap-keys/remap/files/6790676/console.txt)

その他必要な情報がありましたらお知らせください。よろしくお願いいたします。

mokesura, 2021-07-09:
レイアウトを「Right Hand」に変更したときの、Chromeでの「Console」タブのメッセージはこちらです。

> index.ts:18 
> It looks like you're using the development build of the Firebase JS SDK.
> When deploying Firebase apps to production, it is advisable to only import
> the individual SDK components you intend to use.
> 
> For the module builds, these are available in the following manner
> (replace <PACKAGE> with the name of a component - i.e. auth, database, etc):
> 
> CommonJS Modules:
> const firebase = require('firebase/app');
> require('firebase/<PACKAGE>');
> 
> ES Modules:
> import firebase from 'firebase/app';
> import 'firebase/<PACKAGE>';
> 
> Typescript:
> import firebase from 'firebase/app';
> import 'firebase/<PACKAGE>';
> 
> (anonymous) @ index.ts:18
> Configure.tsx:124 componentDidUpdate: false
> Firebase.ts:169 Keyboard definition not found: 65261:52224:CC Proto
> (anonymous) @ Firebase.ts:169
> index.ts:197 strict mode: "items" is 1-tuple, but minItems or maxItems/additionalItems are not specified or different
> J @ index.ts:197
> 2index.ts:197 strict mode: "items" is 2-tuple, but minItems or maxItems/additionalItems are not specified or different
> J @ index.ts:197
> ArrayUtils.ts:2 [Send data]
> ArrayUtils.ts:30 1100 0000 0000 0000 0000 0000 0000 0000  ................
> 0000 0000 0000 0000 0000 0000 0000 0000  ................
> 
> ArrayUtils.ts:2 [Received data]
> ArrayUtils.ts:30 1104 0000 0000 0000 0000 0000 0000 0000  ................
> 0000 0000 0000 0000 0000 0000 0000 0000  ................
> 
> ArrayUtils.ts:2 [Send data]
> ArrayUtils.ts:30 1200 001C 0000 0000 0000 0000 0000 0000  ................
> 0000 0000 0000 0000 0000 0000 0000 0000  ................
> 
> ArrayUtils.ts:2 [Received data]
> ArrayUtils.ts:30 1200 001C 0004 0005 0006 0007 0000 0008  ................
> 0009 000A 000B 0000 001E 001F 0020 0021  ............. .!
> 
> ArrayUtils.ts:2 [Send data]
> ArrayUtils.ts:30 1200 1C0C 0000 0000 0000 0000 0000 0000  ................
> 0000 0000 0000 0000 0000 0000 0000 0000  ................
> 
> ArrayUtils.ts:2 [Received data]
> ArrayUtils.ts:30 1200 1C0C 5CDE 0023 0024 0025 0026 0027  ....\..#.$.%.&.'
> 0000 0000 0000 0000 0000 0000 0000 0000  ................
> 
> ArrayUtils.ts:2 [Send data]
> ArrayUtils.ts:30 1200 281C 0000 0000 0000 0000 0000 0000  ..(.............
> 0000 0000 0000 0000 0000 0000 0000 0000  ................
> 
> ArrayUtils.ts:2 [Received data]
> ArrayUtils.ts:30 1200 281C 0001 0001 0001 0001 0000 0001  ..(.............
> 0001 0001 0001 0000 0001 0001 0001 0001  ................
> 
> ArrayUtils.ts:2 [Send data]
> ArrayUtils.ts:30 1200 440C 0000 0000 0000 0000 0000 0000  ..D.............
> 0000 0000 0000 0000 0000 0000 0000 0000  ................
> 
> ArrayUtils.ts:2 [Received data]
> ArrayUtils.ts:30 1200 440C 0001 0001 0001 0001 0001 0001  ..D.............
> 0000 0000 0000 0000 0000 0000 0000 0000  ................
> 
[... 107 lines cut ...]
> ArrayUtils.ts:30 1200 940C 0000 0000 0000 0000 0000 0000  ................
> 0000 0000 0000 0000 0000 0000 0000 0000  ................
> 
> ArrayUtils.ts:2 [Received data]
> ArrayUtils.ts:30 1200 940C 0001 0001 0001 0001 0001 0001  ................
> 0000 0000 0000 0000 0000 0000 0000 0000  ................
> 
> Configure.tsx:124 componentDidUpdate: false
> ArrayUtils.ts:2 [Send data]
> ArrayUtils.ts:30 0302 0000 0000 0000 0000 0000 0000 0000  ................
> 0000 0000 0000 0000 0000 0000 0000 0000  ................
> 
> ArrayUtils.ts:2 [Received data]
> ArrayUtils.ts:30 0302 0000 0000 0000 0000 0000 0000 0000  ................
> 0000 0000 0000 0000 0000 0000 0000 0000  ................
> 
> ArrayUtils.ts:2 [Send data]
> ArrayUtils.ts:30 0302 0000 0001 0000 0000 0000 0000 0000  ................
> 0000 0000 0000 0000 0000 0000 0000 0000  ................
> 
> ArrayUtils.ts:2 [Received data]
> ArrayUtils.ts:30 0302 0000 0001 0000 0000 0000 0000 0000  ................
> 0000 0000 0000 0000 0000 0000 0000 0000  ................
> 
> DevTools failed to load source map: Could not load content for chrome-extension://iihkglbebihpaflfihhkfmpabjgdpnol/js/script.js.map: HTTP error: status code 404, net::ERR_UNKNOWN_URL_SCHEME

yoichiro, 2021-07-12:
@mokesura Thank you for the reporting. I was able to reproduce this issue. I try to investigate and fix this issue.

yoichiro, 2021-07-14:
@mokesura We have released a new revision to fix this issue. Could you check whether this issue is fixed or not on your environment? Thank you for the feedback!

フィードバックありがとうございます！この問題を解決するための新しいリビジョンをリリースしました。お使いの環境にてこの問題が修正されたかどうか、確認していただくことは可能でしょうか？

closed 2021-07-14 by yoichiro

mokesura, 2021-07-14:
I have confirmed that the problem has been resolved on my keyboard.
Thank you for resolving the issue.

yoichiro, 2021-07-14:
@mokesura Good news for us. Thank you for the confirmation!

PR #495 (2021-07-14): Fix the issue which can't operate the toolbar icons at specific keyboard layout option
This pull request is to fix #493.

When specifying some keyboard layout option, the toolbar icons can't be operated (for instance, we can't click them). I investigated this issue, then I found the cause of this issue.

[image]

In the target JSON file and specifying the keyboard layout option, the `left` value of the `keyboard-frame` class becomes non-zero. However, the width of the class is the same as other normal keyboard layout option. As the result, the pane with the `keyboard-frame` class is put over t [... cut]