remap-keys/remap #494: [Layout] keymap crooked

3araht, 2021-07-11:
As shown below, the 'floating' keys are misaligned, and the other keymaps are also shown incorrectly.
It affects the other layers.

This is the keymap how it should be, drawn by [Keyboard Layout Editor](http://www.keyboard-layout-editor.com/).
[image]

And this is how it was shown in REMAP.
[image]

The "Layout Option" doesn't work.

Other layers are affected, as well. Here's the code, 
[image]

And here's the cheat sheet.
[keymap_cheatsheet_chromatonemini.pdf](https://github.com/remap-keys/remap/files/6796902/keymap_cheatsheet_chromatonemini.pdf)

3araht, 2021-07-11:
[Here's](https://github.com/3araht/chromatonemini/blob/main/temp/qmk_firmware/keyboards/chromatonemini/chromatonemini_via.json) the json file.

yoichiro, 2021-07-11:
@3araht Thank you for the report! Could you also submit your hex file? We will be able to reproduce this issue more easily on our local environment.

3araht, 2021-07-12:
Here you are.
https://github.com/3araht/chromatonemini/blob/main/chromatonemini_via_hex.zip

yoichiro, 2021-07-12:
@3araht Thank you for the submitting the hex file! We will investigate this issue and will send feedback after that.

yoichiro, 2021-07-13:
@adamrocker Could you investigate this issue?

adamrocker, 2021-07-16:
I've investigated this issue and found a problem with the definition of the key's matrix in the JSON.

yoichiro, 2021-07-16:
@adamrocker Thank you for the investigating!

@3araht I think that you have already updated the keyboard definition JSON file to fix this issue. I close this issue, but you can reopen this issue anytime, when you find other issues regarding this.. Thank you for your contribution!

closed 2021-07-16 by yoichiro