remap-keys/remap #504: Can't open keyboard when using USB Hub

Pekaso, 2021-07-17:
- Environment
    - OS: Windows10 Pro 21H1 build 19043.1083
    - Browser: Google Chrome 91.0.4472.124（Official Build） （64 ビット）
    - USB Hub: [USB-2H1001BK](https://www.amazon.co.jp/dp/B01K3EGG46)
        - It has 3 USB Hub ICs "Teaminus Tech. FE1.1s". These are cascade connection.
        - VID_1A40&PID_0101#
    - Keyboard: ALETH54

- Issue
    - After connecting the keyboard to Remap, it is stuck at "Opening keyboard".
    - The number of nodes in the USB hub is not related.
    - HID connection permission dialog appears and I can connect to browser by the normal operation.

[image]
[image]
[remap-keys.app-1626509037508.log](https://github.com/remap-keys/remap/files/6834387/remap-keys.app-1626509037508.log)

yoichiro, 2021-07-17:
@Pekaso Thank you for the detail report! I intend to try to reproduce this issue and investigate this reason.

yoichiro, 2023-09-12:
We think that we don't have any solution to solve this issue. Close this.

closed 2023-09-12 by yoichiro