remap-keys/remap #510: Keyboard catalog register: Don't refresh the display after dragging an image

marksard, 2021-07-28:
Dragging an image file does not update the display of the image the first time.
[image]

If go to the remap top page, go to the registration page again, and drag the same image again, it will succeed.

yoichiro, 2021-07-29:
@marksard Thank you for the feedback. I heard this issue from other person. I intend to confirm and fix this issue asap.

yoichiro, 2021-08-10:
I can't reproduce this issue on my environment...

mrnoisytiger, 2022-07-05:
@yoichiro I sometimes have this issue too. I will drag an image onto that page, and it will stall on 100% for about 1 minute. Then, the page will refresh and the image upload will appear to be successful. 

Microsoft Edge: Version 103.0.1264.44 (Official build) (x86_64)
MacOS Catalina 10.15.7

mrnoisytiger, 2022-07-05:
https://user-images.githubusercontent.com/12619586/177257809-133e74c1-f4e7-44c0-bffe-db2dcb66b37d.mov

mrnoisytiger, 2022-07-05:
[image]

The following are the errors I receive upon the first upload. A refresh will fix this.

mrnoisytiger, 2022-07-05:
[image]

Note the following errors upon uploading a sub-image. The image uploads successfully after the main image.