remap-keys/remap #532: Alt or Ctrl "shifted" characters are not visible in the editor

Silverquark, 2021-08-03:
In a lot of languages there are special keycombination with the right alt key or the ctrl key. 

For example in the german qwertz layout the {}[]€@~ symbols requires the user to press the alt gr (right alt) key + another character

[image]

yoichiro, 2021-08-04:
@Silverquark After deployed #530, I understand that this issue was already fixed. Is this correct?

Silverquark, 2021-08-04:
No, the one i fixed was the symbols while Holding the shift key. In qwertz for example the 7 key is used for ")" when Holding shift and also for "[" when Holding the right alt key (alt gr).
It appears that someone started implementing this, but i dont think it was finished, as i tried using it, but it didnt work.
I planned in talking a look into fixing it, but havent found the time for it yet

yoichiro, 2021-08-04:
@adamrocker Could you catch up this issue?

adamrocker, 2021-08-23:
@yoichiro Sorry for the late reply. I fixed the bug for this.
Could you please review my PR?
https://github.com/remap-keys/remap/pull/574

closed 2021-08-23 by yoichiro