remap-keys/remap #533: [Feature request] キーボードカタログの表示件数を増やす

ShinoharaTa, 2021-08-04:
現在のキーボードカタログの画面は表示件数が5件のみで、ページネーションが都度必要です。
カタログのメリットを考えると、やはりその一覧能力だと思っておりまして、もっと大量に表示してほしいと思うことがあります。

検索条件を入れる前のページではランダム表示で新しいキーボードとの出会いを期待したいです。

yoichiro, 2021-08-04:
@ShinoharaTa Thank you for the suggestion. To increase the number of search result list, do you have an idea about how to show the list? 提案ありがとうございます。検索結果のリストの件数を増加させるために、そのリストをどのように表示するか、何かアイディアをお持ちでしょうか？

yoichiro, 2021-08-04:
> 検索条件を入れる前のページではランダム表示で新しいキーボードとの出会いを期待したいです。

@ShinoharaTa We have already implemented the random sort order logic. Therefore, the listed keyboards are listed as the random sort without any searching conditions. 既にランダムでの順序表示ロジックは実装されています。つまり、検索条件なしでは、キーボードのリストはランダムでソートされています。

ShinoharaTa, 2021-08-10:
返事が遅くなりました。
ランダムのロジックは既に実装されているとのことで、了解です。

今は、写真、タイトル、作者名がRowで並んでいます。余白が非常に多いので、4カラムor 5カラムで10行ほど（40件ぐらい）並べてもいいかもしれません。
一つの画面に、大量に表示されているところからゆっくり見たいです。

yoichiro, 2021-08-10:
@ShinoharaTa ご意見ありがとうございます。

> 一つの画面に、大量に表示されているところからゆっくり見たいです。

現状の表示が好ましいとは考えていませんので、何らかの改善をしていきたいと考えています。その方向性として、仰るとおりに1ページあたりの表示件数を多くする、という方向性と、検索結果1つずつにもっと情報量を多くする、という方向性があります。

おそらく @ShinoharaTa さんのアイディアでは、キーボードの画像を主としてそれを並べて比較する、ということかと想像します。それとは別に、検索条件として「どのような機能を有しているか」という条件指定を採用しているため、検索結果にて機能差での比較ができたほうが望ましいだろう、ということも求められています（そういったフィードバックを実際にもらっています）。

検索結果の表示方式は複数提供されていて、ユーザが好みの方式を選択できる、が好ましい状態かなと思いますので、「現状のシンプルなリスト」と「画像を並べた件数の多いリスト」でまずは検討してみようかなと思います。

yoichiro, 2021-08-10:
@ShinoharaTa もう一つ表示数を増やしにくい理由として、現状でもキーボード名で最も長いものは30文字を超えていて、そのための表示領域を確保しないといけない、ということがあります。

yoichiro, 2021-08-10:
Probably, he expected the list like below... I guess.

[image]

ShinoharaTa, 2021-08-10:
確かにキーボード名や作者名の長さはカラム表示では考慮すべきポイントとなってしまいますね。
現状では1ページに5台までで、以降はページネーションにて都度表示しています。
私は20件でも、またはそれ以上でも載っていていいとおもいますが、この件数なども、やはりユーザー側で選択できるのが好ましいのかもですね