remap-keys/remap #548: JIS Zenkaku/Hankaku がlang5を送出している

soogie, 2021-08-08:
Hold/TapのTap欄の候補には「半角/全角」が出てこず，選択できるのはJIS Zenkaku/Hankakuです。

そこでこのように設定しました。
[image]
[image]

しかしWindows機につないでこのキーをタップしても半角/全角キーとして機能しません。

あらためてMacに繋ぎ直してkaravinar event viewerで送出されるキーを確認したところ，lang5が送出されていました。
[image]

正しくはKC_GRVを送出し，こうなるはずだと思います。
[image]

なお，単体で「半角/全角」をドラッグ＆ドロップした際は問題ありませんがKey欄でJIS Zenkaku/Hankakuを選択すると同様の挙動となります。

ご確認いただけますでしょうか。

yoichiro, 2021-08-08:
@soogie Issue登録ありがとうございます。状況を確認いたしました。複数の要因が重なった結果、ご報告の状況となった模様です。

まず、`KC_LANG5` の Label として `JIS Zenkaku/Hankaku` と表示されていますが、これは QMK Firmware の [Full Keycode List](https://docs.qmk.fm/#/keycodes) に基づいて設定された結果です。ただし、これは実際には「半角/全角」キーを反映したものではないと考えられますので、修正の必要がありそうです（ `Language 5` に修正したほうが良さそうです）。

次に、ご指摘通り、`KC_GRAVE` のキーコード 0x35 が実際には「半角/全角」キーとなります。キーボードの言語を「Japanese」にした状態で、「KEY」タブで keycode に「半角」と入力すると、「半角/全角」が候補に出てきます。これは正しい動作かと思います。

[image]

しかし、「HOLD/TAP」タブにて「TAP」に「半角」と入力しても、確かに上記のように「半角/全角」が候補に出てきません。

[image]

これは、おそらくキーボードの言語設定を変更した際に本来適用されるべき言語ごとのラベルが、HOLD/TAP タブ上にて考慮されていないのではないか、と推測されます。

@adamrocker 上記、確認できますでしょうか？

yoichiro, 2021-08-08:
This issue occurs for Japanese keyboard only, therefore we're using Japanese to discuss about this issue.

soogie, 2021-08-09:
Hold/Tapについては現象は解消しました。
[image]

単体キーについても何かアサインされている場合は問題なさそうです。
ただ，NOOPに割り当てたキーではKeycode欄の候補がおかしくなっています。
半角/全角が出てこないだけでなく，変換無変換やアルファベットのAすらも候補に出てきません。
[image]

この状態でKeycode欄にAと入力してもこのようにアルファベットAが候補に出ません。
[image]

なお，ここからx(clear)をクリックして入力をキャンセルすると，全ての候補が正しく表示される状態になります。
[image]

NOOP以外のキーの場合，最初にx(clear)をクリックするがNOOPでは最初から候補が出る，というところの違いが関係しているように思います。

yoichiro, 2021-08-09:
@soogie ご報告いただいた挙動については「仕様通り」となります。NOOP（KC_NO） が割り当てられている状態では、KEY タブの Keycode フィールドには ` ` （空白一文字）が入力されています。その状態で補完候補一覧を表示した場合には「空白を含む候補の一覧」が表示されますので、例えば「A」については空白が含まれていませんので補完候補には出てきません。また、「半角/全角」についても空白が含まれませんので、同じく補完候補には含まれません。そして、NOOP を示す空白文字を削除する、もしくは X をクリックして同じく入力された文字を削除することで、全ての候補が表示されます。

PR #549 (2021-08-08): Fix the issue #548.
Fix #548.

This pull request has two modifications:

(1) Currently, the `JIS Zenkaku/Hankaku` is assigned to the `Lang 5`. However, actually the `半角/全角` key code is 0x35 (this is `KC_GRAV`). Therefore, I think that we should remove the `JIS Zenkaku/Hankaku` label from the `Lang 5` key.

[image]

(2) The keyboard language setting is not passed to the `TabHoldTapKey` component (instead, `en-us` is passed), therefore the `半角/全角` key label is not shown on the HOLD/TAP tab panel when the `Japanese` is selected.  The `props.labelLang` value should be passed to the `TabHoldTapKey` [... cut]
closed 2021-08-09 by yoichiro