remap-keys/remap #550: [Featyre

hsgw, 2021-08-09:


yoichiro, 2021-08-10:
@hsgw Is what you expect a JSON file exported from a VIA app with "Save Current Layout"? Or, is that actual C code?

hsgw, 2021-08-10:
I assumed it would be something similar to export from VIA.

yoichiro, 2021-08-18:
@hsgw Could you describe some use cases about the feature you suggested?

yoichiro, 2021-08-18:
@hsgw Thank you for the description. However, Remap is providing the Keymap Save/Restore feature to save multiple key mappings to the Remap database and to restore them. That is, users can persistent their key mappings by using the feature without downloading a file which has a key mapping.

What is the pros of your idea better than the save/restore feature?

hsgw, 2021-08-19:
@yoichiro I think the save/restore function is not enough as far as the keymaps are only on EEPROM. As I mentioned in my previous post, EEPROM can be easily cleared.
Of course, the save/restore feature is useful, but it is inconvenient to use remap to restore keymap.

Also, it's useful for making the default keymap during development.

adamrocker, 2021-08-19:
@hsgw Sorry, I'm confusing the situation you mentioned.
Let me clarify my understand.
 
Do you mean, you do NOT want to use Remap to restore the keymap you customized using Remap?

hsgw, 2021-08-19:
> Do you mean, you do NOT want to use Remap to restore the keymap you customized using Remap?

Partially yes. I think it's too much trouble.
And, now I'm using VIA to export the keymap.

As a confirmation, there is no way to download the keymap locally in remap, right?

adamrocker, 2021-08-20:
> As a confirmation, there is no way to download the keymap locally in remap, right?

Yes. Remap does not provide a download feature, now.

> Partially yes. I think it's too much trouble.
> And, now I'm using VIA to export the keymap.

Interesting! That's good news.
It's easy for you to customize a keymap using Remap. That's our goal, now.
And I could understand the background of your feature request is that it might be useful for keyboard designers/developers, like you, during keyboard designing/development.

Actually, the current save/restore keymap feature of Remap is for keyboard users, not keyboard designers/developers.
I think it's quite an interesting idea for expanding Remap features for designers/developers.

Anyway, I could understand that this request is for keyboard designers/developers.

@yoichiro 
We should keep considering the Remap roadmap not only for users but also for designers/developers, I think.
What do you say?

yoichiro, 2021-08-20:
@adamrocker Yes, I agree your idea. We should consider a roadmap/features for keyboard designers/developers in the future, but unfortunatelly I think it is not now. We have still some plans to provide features for keyboard users. Also, we should design a feature set for keyboard designers/developers comprehensively. I guess that this proposal might be one of them or not.

hsgw, 2021-08-20:
I understood about your roadmap. I hope it goes well👍
Thank you!

closed 2021-08-20 by hsgw

gshmu, 2022-11-12:
I want update my keymap.c default keymap. 
That will make my tuning my keyboard smoother, instead of apply keymap after flash.  thank you

mxmilkiib, 2024-06-06:
Could it be possible to reopen this issue as a placeholder for hopefully eventually exporing either the .c and .h/full firmware source, or a keymap JSON for QMK Configurator/VIA? Thanks!

If it's a QMK Configurator Keymap then the Configurator [etc.](https://jhelvy.shinyapps.io/qmkjsonconverter/) can export the source for playing about with in the terminal, so certain compex macros can be added (VIA can help with some macros but not all, and I'm not keen on the VIA interface).