remap-keys/remap #558: [Feature Request] キーマップの引っ越し機能

flat35hd99, 2021-08-15:
## 背景

corneとcorneliusは物理配列が似通っているため、そのまま移して利用するために手作業でcorneliusのキーマップを作製しました。しかし、各レイヤーの配列をほとんどそのままコピーしなくても済む方法があったらなあと思いました。

## 方法

1. remapに登録された二つのキーボードの物理配列情報と、移行元のキーマップを用意します。
2. 移行元から移行先への物理配列の対応をユーザーが定義します。
3. ユーザーが定義した対応から、各レイヤーのキーマップを移行先のキーマップとして作成します。
4. アプリ側に保存します。
5. ユーザーは移行先のキーボードをremapに接続し、作成・保存されたキーマップを読み込みます。

yoichiro, 2021-08-15:
@flat35hd99 Thank you for the filing this issue.

yoichiro, 2021-08-15:
I'm not sure how many users need this use case. Basically, most keyboards have each different keyboard layout, therefore, the applying a key mapping to other keyboard is not easy. Also, the number of combinations of keyboards are so lerge.

Currently, I don't want this feature, but please post :+1 if you want this feature.