remap-keys/remap #561: Set focus to the keycode input field at opening the custom key popup UI.

yoichiro, 2021-08-16:
As the title. If the keycode input field is focused after opening the popup, users can modify the keycode immediately.

elfmimi, 2021-08-17:
I've filed https://github.com/remap-keys/remap/issues/562 . effectively down voting this one.

yoichiro, 2021-08-17:
Fom #562 

> Currently, it takes tapping TAB key for twice to make the keycode field focused.
I think this can be reduced to once, by giving the "KEY" tab focus upon opening the popup.
>
> To explain what I'm saying clearly, please try the following steps.
> 
> open the popup UI by clicking on any element representing a key.
> then click on the "KEY" tab. this makes the tab element focused.
> press TAB key to move focus to the keycode field.
> What I want is above behavior but without doing step 2 manually.

elfmimi, 2021-08-18:
webというメディアに求められる特性に誰にでも使いやすい(ユニバーサルデザイン)というのが、あると思うのですが、Remapは、webの基本に忠実に作られているためもともとそれが叶っているように思います。
ポップアップが開いた時点でkeycodeが全選択状態であると、ゲーマー的に最小アクション数を求める向きには歓迎されると思いますが万人向けではないかも知れないと思ったので562の方を提案しました。個人的にはどっちを採用してくださっても構わないです。
 
I recognize one of the mandatory characteristics of web media is ease of use for everyone, i.e. universal design.
and I'd say, because Remap is carefully made with basics of the web in mind, it inherently possess that quality.
That said, if the keycode field is in selected state when the popup is opened, gamers and folks who prefer minimum actions possible will welcome it, but it might not be preferable for everyone because it could feel like slightly intimidating UI. so I suggested #562. personally, whichever choice is okay.

yoichiro, 2021-08-19:
@elfmimi To become more user-friendliness, we decided to apply that the input field is focused automatically and the inputted text is selected when opening the custom key popover UI. We intend to watch whether users can operate a key assigning easily with the popover UI. 利便性を考慮して、カスタムキーポップアップUIが開く際に、入力フィールドをフォーカスして入力値を選択状態することを適用します。そのポップアップUIにてユーザがより簡単にキーアサインを操作できるかどうか様子を見ていく予定です。

closed 2021-08-19 by yoichiro