remap-keys/remap #589: Keyboard layout is not displayed (Browser gets whiteout)

jpskenn, 2021-08-27:
After connecting HID device, keyboard layout is not displayed(Browser gets whiteout).
This happens with "[Jones](https://remap-keys.app/catalog/z8HynboRCOQfPmwXbzp6)" keyboard.

For a moment, Header bar(Keyboard name, ID, etc.) and messages(*) displayed, then whiteout.

*messages:
- Fetching  keyboard definition.
- Opening keyboard.

Chrome DevTools Console log:
[remap-keys.app-1630106979687.log](https://github.com/remap-keys/remap/files/7069811/remap-keys.app-1630106979687.log)

yoichiro, 2021-08-28:
@adamrocker We can reproduce this issue by accessing to the following URL: https://remap-keys.app/catalog/z8HynboRCOQfPmwXbzp6/keymap

When reproducing this issue, I got the following output in the console tab of the Chrome DevTools:

```
react-dom.production.min.js:216 TypeError: Cannot assign to read only property 'd' of object '#<Object>'
    at Function.assign (<anonymous>)
    at new e (KeyModel.ts:58)
    at KeyboardModel.ts:328
    at Array.forEach (<anonymous>)
    at e.value (KeyboardModel.ts:327)
    at new e (KeyboardModel.ts:164)
    at a.value (Catalogkeymap.tsx:101)
    at Ua (react-dom.production.min.js:187)
    at Fa (react-dom.production.min.js:186)
    at Vs (react-dom.production.min.js:269)
```

Probably, I guess that this issue was caused by the pull request #584. Could you confirm this issue?

yoichiro, 2021-08-28:
@jpskenn Thank you for the reporting! We fixed this issue and now deploying the fixed version. Please confirm whether this issue has already been fixed or not.

closed 2021-08-28 by yoichiro

jpskenn, 2021-08-28:
Confirmed the issue fixed on "Build 125"👍
Thank you.

PR #590 (2021-08-28): Fix the issue that assigning a value to the readonly property.
Fix #589 

The `keyOp` property is a readonly property, but the `Object.assign` function tries replacing the value and it is filed. As the result, the keyboard layout is not shown.

Instead of using the `Object.assign`, use the spread expression to avoid the assigning the value to the readonly property.