remap-keys/remap #608: Lightning support for RGB_Matrix

ToastyStoemp, 2021-09-12:
Via currently does not support RGB_Matrix lightning controls
(you can assign the keycodes to switches, but can't directly set the animation and colour from the app)

Would be a huge feature if this was possible through Remap

yoichiro, 2021-09-12:
@ToastyStoemp We recognize that Remap already have such lightning control feature like the following:

[image]

Was the feature you mentioned like the above?

ToastyStoemp, 2021-09-12:
This control is only for RGB lightning,
Adding control for [RGB Matrix lightning](https://beta.docs.qmk.fm/using-qmk/hardware-features/lighting/feature_rgb_matrix) is what I'm hoping for.

RGB Matrix allows you to map LED's to specific switches, example from the KeyBee65 PCB: 
[image]

RGB Matrix also allows you to have per key LED and underglow in one chain of individually addressable led's.

ToastyStoemp, 2021-09-12:
Mock-up
[image]

yoichiro, 2021-09-12:
@ToastyStoemp Ah, I see. Yes, the RGB lightning and RGB Matrix are different. But, unfortunately, I guess that we cannot implement such feature currently. Because, Remap uses the VIA protocol to implement each feature, but I recognize that the VIA protocol does not support the RGB matrix.

ToastyStoemp, 2021-09-12:
Ah so it's not something we could add without via supporting it

JohnAZoidberg, 2023-02-01:
It's supported now: https://github.com/qmk/qmk_firmware/blob/master/quantum/via.c#L637
Since: https://github.com/qmk/qmk_firmware/pull/16086