remap-keys/remap #612: Multiple keys are selected when changing a layout option

yoichiro, 2021-09-13:
After changing the layout option of the Attack25 keyboard, multiple keys are selected.

[image]

Source: https://twitter.com/checkLa1n/status/1436851102004101123

adamrocker, 2021-09-13:
I found some issues with the keyboard definition file(.json).
There are 2 same matrix index;'1,3' and '3,4', whenthose options are activated.
Is it possible to fix those matrix indexes as below?
[image]

yoichiro, 2021-09-13:
@adamrocker Thank you for the investigating!

@monksoffunk We found some issues in the keyboard definition JSON file of your Attack25 keyboard. Could you confirm them and fix it? It would be happy if you update the JSON file into Remap after fixing it.

monksoffunk, 2021-09-13:
Thank you for reporting the bug.
Already updated and I think working correctly now.

yoichiro, 2021-09-13:
@adamrocker @monksoffunk Thanks. I close this issue.

closed 2021-09-13 by yoichiro