remap-keys/remap #636: Missing Magic Keycodes

jpskenn, 2021-09-16:
Some [Magic Keycodes](https://docs.qmk.fm/#/keycodes_magic) are missing.

- LCG_SWP
- LCG_NRM
- RCG_SWP
- RCG_NRM
- EH_LEFT
- EH_RGHT

closed 2021-09-17 by yoichiro

yoichiro, 2021-09-17:
@jpskenn These key codes have been added into the device category. Could you confirm them? Thank you for the feedback!

[image]

jpskenn, 2021-09-17:
Thanks to add key codes.
All key codes recognized as normal👍

I checked with this code.

```c
bool process_record_user(uint16_t keycode, keyrecord_t *record) {
...
...
        case VERSION: // Output firmware info.
        case EH_LEFT:
        case EH_RGHT:
        case LCG_SWP:
        case LCG_NRM:
        case RCG_SWP:
        case RCG_NRM:
        case CG_NORM:
        case CG_TOGG:
        case CG_SWAP:
            if (record->event.pressed) {
                SEND_STRING (QMK_KEYBOARD ":" QMK_KEYMAP " @ " QMK_VERSION " | " QMK_BUILDDATE);
            }
            return false;
        default:
            break;
    }
    return true;
```