remap-keys/remap #640: Layout Option is not working appropriately.

3araht, 2021-09-20:
giabalanai keyboard on REMAP has an option to show/hide the keys assigned to a rotary encoder.
It used to work correctly. However, on Sep 20th, 2021, I found that the option was not working correctly.
The keys are shown on the upper left corner, overlapping the keys on the left side keyboard.

Those have to be shown on the upper right side, as shown below.
[image]

This is the layout without keys for rotary encorder.
[image]

Currently, it overlaps as shown below.
[image]

Here's the json file.
[giabalanai_default_via_json.txt](https://github.com/remap-keys/remap/files/7195504/giabalanai_default_via_json.txt)

adamrocker, 2021-09-21:
Thank you for your report. I'm gonna check it :)

adamrocker, 2021-09-21:
@3araht I'm sorry. I misunderstood the layout option definition.
According to [the VIA definition document](https://caniusevia.com/docs/layouts/#:~:text=Layout%20option%20choices%20must%20all%20have%20the%20same%20coverage%2C%20i.e.%20they%20overlap%20exactly%20in%20shape%2C%20but%20can%20vary%20in%20keycap%20sizes%20and%20arrangement.), if the keyboard layout options are set, it must have the default option.

In this case, Remap must NOT show the 6,8 and 6,9 keys anywhere even if the option choice 1 is ON because there is not any default options.

If you want to switch the visibility of the 6,8 and 6,9 keys, you can consider using decal keys with default options like `{d: true},6,8\n\n\n0,0`.
Please refer the KLE idea I made.
http://www.keyboard-layout-editor.com/##@@=0,0&=1,0&=2,0&=3,0&=4,0&=5,0&=5,5&=4,5&=3,5&=2,5&=1,5&=0,5%3B&@_x:0.5%3B&=0,1&=1,1&=2,1&=3,1&=4,1&=5,1&=5,6&=4,6&=3,6&=2,6&=1,6&=0,6&_x:2%3B&=8,0&_x:11.5%3B&_d:true%3B&=6,8%0A%0A%0A0,0&_x:1%3B&=6,8%0A%0A%0A0,1&=6,9%0A%0A%0A0,1%3B&@_x:1%3B&=0,2&=1,2&=2,2&=3,2&=4,2&=5,2&=5,7&=4,7&=3,7&=2,7&=1,7&=0,7&_x:2%3B&=6,0&=6,1&=6,2&=6,3&=6,4&=6,5&=6,6&=9,6&=9,5&=9,4&=9,3&=9,2&_x:0.5&c=%23777777%3B&=9,1%3B&@_x:1.5&c=%23cccccc%3B&=0,3&=1,3&=2,3&=3,3&=4,3&=5,3&=5,8&=4,8&=3,8&=2,8&=1,8&=0,8&_x:1%3B&=7,0&=7,1&=7,2&=7,3&=7,4&=7,5&=7,6&=10,6&=10,5&=10,4&=10,3&=10,2&=10,1%3B&@_x:2%3B&=0,4&=1,4&=2,4&=3,4&=4,4&=5,4&=5,9&=4,9&=3,9&=2,9&=1,9&=0,9&_x:1%3B&=8,1&=8,2&=8,3&=8,4&=8,5&=8,6&=11,6&=11,5&=11,4&=11,3&=11,2&=11,1

closed 2021-09-21 by yoichiro