remap-keys/remap #651: Unable to change keymap in chrome on manjaro linux

m3y, 2021-10-03:
## Contect
I try to change the keymap in the following environment, but "Could not open" is displayed and I cannot change it.

## Environment
OS: Manjaro linux
Brawser: Chrome(94.0.4606.71)
keyboard: Cornelius
```
Product Name Yushakobo Cornelius
Vendor ID        0x3265
Product ID       0x0005
```

yoichiro, 2021-10-03:
@m3y You need to change the permission of the target device file for the keyboard you want to manipulate on Remap.

closed 2021-10-03 by yoichiro

perepicornell, 2021-10-13:
Thanks @yoichiro for leading me here.
For further people with this issue, some info:

The device's files in my linux (ubuntu) were located at `/dev/input`.
At /dev/input/by-id I could locate which file in particular was the right one.

Then I did:
`sudo setfacl -m u:your_user_name:rw /dev/input/eventX`

Apparently the permissions are set on startup so you might need to set this everytime you want to use the remap app.

nomm-dot, 2024-07-03:
> I encountered the same issue with my Keyball44, and resolved by creating a file `/etc/udev/rules.d/50-keyball.rules` with contents
> 
> ```
> # Keyball44
> KERNEL=="hidraw*", SUBSYSTEM=="hidraw", ATTRS{idVendor}=="5957", ATTRS{idProduct}=="0400", MODE="0666"
> ```
> 
> , reload udev with `udevadm control --reload`, and reconnect the device. The configuration persists, so you don't have to run commands every time you want to remap keys. Though this won't work until you flash the firmware for the first time, as it uses the vendor/product to specify the device.

This worked in nixos as well, I added the following to my configuration.nix for my keyball39, ran `sudo nixos-rebuild switch`, and rebooted. Worked great, thanks for the help!
```
  services.udev.extraRules = ''
    KERNEL=="hidraw*", SUBSYSTEM=="hidraw", ATTRS{idVendor}=="5957", ATTRS{idProduct}=="0200", MODE="0666"
  '';
```