remap-keys/remap #676: [Layer] unable to select layers when left side keys are hidden.

3araht, 2021-12-03:
When left side keys are hidden, layers can't be selected.
When the hidden keys on the left side are shown, there is no problem.

It happens2 with a giabalanaipico keyboard, which is under review.
[image]

[image]

yoichiro, 2021-12-05:
@3araht I have changed the code to fix this issue and have deployed it. Could you confirm whether this issue is fixed or not?

3araht, 2021-12-05:
> @3araht I have changed the code to fix this issue and have deployed it. Could you confirm whether this issue is fixed or not?

Nope. Nothing seemed to be changed. Tested at 13:35 on Dec 5th.

yoichiro, 2021-12-05:
@3araht Thank you for the confirmation. Ok, I intend to test with my Raspberry Pi Pico.

PR #679 (2021-12-05): Fix #676. Add the z-index style to be able to click the layer UI.
Fix #676

The keyboard layout UI and the toolber UI have the style `z-index:2`, but the layer UI doesn't have the style. The reason of the issue #676 may be the nothing of the z-index style. This pull request adds the z-index style to the layer UI to fix the issue.
commit 137c8725 (2021-12-05): Fix #676. Add the z-index style to be able to click the layer UI.
commit de39a941 (2021-12-05): Merge pull request #679 from remap-keys/fix-676

Fix #676. Add the z-index style to be able to click the layer UI.