remap-keys/remap #692: Can't update a key with DnD

yoichiro, 2022-01-12:
Currently we can't update a key with DnD as the following movie:

https://user-images.githubusercontent.com/261787/149048339-42823442-3559-4010-91e4-266838d47f37.mp4

yoichiro, 2022-01-12:
@adamrocker Could you confirm and fix this issue?

adamrocker, 2022-01-12:
@yoichiro I'm gonna check it :)

yoichiro, 2022-01-12:
@adamrocker Thank you for the confirmation! It seems that this issue occurs on Linux Chrome only. I investigate this issue.

adamrocker, 2022-01-12:
@yoichiro Thank you so much. It works fine with macOS's Chrome as I reported.

yoichiro, 2022-01-12:
I tested this issue on Windows 11 and Chrome OS, but I was not able to reproduce this issue.

yoichiro, 2022-01-12:
I close this issue because this issue might be occur on my Linux environment only.

closed 2022-01-12 by yoichiro