remap-keys/remap #701: [Feature Request] Edit macro with keyboard

sekigon-gonnoc, 2022-03-19:
I want to edit macro with keyboard. 
 Because one of the uses of macro is to register strings, it would be useful to be able to paste text or type directly from the keyboard.
 
 My suggestion is adding these feature to macro editor:
  - Push 'ascii' key to macro buffer if any ascii key is captured in `keydown` events
  - Pop last key from macro buffer if backspace is captured in `keydown` events
  - Push 'ascii' keys converted from clipboard's text on `paste` events
 
 If you are interested, I will submit a pull request.

yoichiro, 2022-03-19:
@sekigon-gonnoc Thank you for this idea! Basically, looks good to me, but I have some questions below:

* Which element does handle keyboard events?
* I guess that it is necessary to highlight the element above so that users can recognize where they can push and pop characters. And, for example, another idea is to put a message to let users know that they can push and pop characters to the key code sequence with typing a keyboard.
* Is this feature implemented with Clipboard API?

sekigon-gonnoc, 2022-03-20:
Thank you for response. Answers are following:

- `.macro-editor-content-keys`
- I changed place holder text to "Please drop a keycode or enter text here!"
- I'm not familiar with Clipboard API but it may be not used. The feature is implemented by using ClipboardEvent.clipboardData

yoichiro, 2022-03-20:
@sekigon-gonnoc Thank you for the answers. LGTM at this time. We will check your implementation on a new pull request from you. Could you go forward?

PR #704 (2022-07-02): Edit macro by keyboard
This pull request enables editing macro with keyboard.

Resolve #701
closed 2022-07-02 by yoichiro