remap-keys/remap #702: [Feature Request] Edit keymap with keyboard

sekigon-gonnoc, 2022-03-19:
I want to edit keymap with keyboard. 

My suggestion is following change for keymap editor:
- `selectedPos` is not cleared when the custom key popup is closed
- Keymap setting is updated on keydown events. New key setting is generated from pressed key.
- Automatically select next key

yoichiro, 2022-03-19:
@sekigon-gonnoc Thank you for this feature request! Do you already have a code set to implement this idea?

sekigon-gonnoc, 2022-03-20:
Yes, at the moment I'm considering detect keyboard events with a div that enclose `Keymap` component.
If you have any other ideas, I will try that way.

https://github.com/remap-keys/remap/commit/c393e45c2ca22b6387b7d980fb3596e4f41282f0

yoichiro, 2022-03-20:
@sekigon-gonnoc Great. Basically, the idea looks good. Could you create a new pull request for this? Thanks!

PR #705 (2022-08-17): Edit keymap with keyboard
This pull request enables editing keymap with keyboard.

Resolves #702
closed 2022-08-17 by yoichiro