remap-keys/remap #713: Empty user name

otahinosame, 2022-06-08:
My user name is not displayed on the screen for saving and sharing keymaps.
When I Save and Share, my user name shows null.

[image]

[image]

yoichiro, 2022-08-18:
@otahinosame I have fixed this issue by changing the logic like the following:

1. If `user.displayName` is not null, use it.
2. If `user.providerData[n].displayName` is not null, use it.
3. If `user.email` is not null, use its user name part.
4. Otherwise, use "(no name)".

closed 2022-08-18 by yoichiro

otahinosame, 2022-08-18:
@yoichiro Thank you. I tried sharing a keymap and it worked well.

yoichiro, 2022-08-19:
@otahinosame Great! Thank you for the feedback.

PR #727 (2022-08-18): Change the logic to determine the display name of the authenticated user
Fix #713 

This pull request changes the logic how to determine the display name of the authenticated user when storing a key mapping data.