remap-keys/remap #718: Layer 1 is not displayed.

telzo2000, 2022-07-13:
At cool650half, layer 1 is not displayed and layer 2 is displayed on layer 1.
Even if I change the key of layer 1 and overwrite it, it is not written.
When the key of layer 2 is changed, the behavior of replacing the key of layer 1 is shown.
I checked the firmware, but I couldn't identify the cause, so I wondered if there was something wrong with the remap side.
thank you.
https://github.com/telzo2000/cool650/tree/main/firmware_cool650_half

[image]
[image]

yoichiro, 2022-08-09:
@telzo2000 Could you confirm whether there is an error message or not in the Console tab in the Chrome DevTools when your situation occurs?

telzo2000, 2022-08-10:
Thank you for reply.
I looked at the Console tab in the Chrome DevTools and realized I was the culprit.
The cause was found to be a description mistake in info.json.
I updated the info.json and the issue was resolved.
Thank you for always helping me.

yoichiro, 2022-08-10:
@telzo2000 Thank you for the confirmation. OK, I close this issue.

closed 2022-08-10 by yoichiro