remap-keys/remap #719: Internationalization

yoichiro, 2022-08-15:
Currently Remap supports English notations only on all pages and UIs. However, it would be difficult to understand them for non-English users (especially as Japanese). If Remap supports other languages, Remap UIs would become more friendly.

yoichiro, 2022-08-15:
Discussion on Discord (Japanese only): https://discordapp.com/channels/777912889344393216/777917309209411654/1008532518343028766

Katsuyuki-Karasawa, 2022-08-15:
```
- [ ] ja-JP
- [ ] ko-KR
- [ ] zh-CN
- [ ] zh-tw
```
I've already seen the discussion, so I made a checkbox for which languages we will support lightly!
I think it would be easier to understand if I put the checkboxes in a folder related to i18n, like `progress.md`.

Also, I am not sure if it could handle `.js`, but since Remap is OSS, [you can use it for free if you apply to Crowdin](https://crowdin.com/page/open-source-project-setup-request), so you may utilize it depending on the file format.

yoichiro, 2022-08-15:
The step I imagined is:

1. Decide the target pages or UIs we want to translate.
2. Decide how to translate them. For example, what library should we use, or should we translate without using any library, where should we put resource files, or etc.
3. Define a resource file for the language which Remap is currently using.
4. Define translated resource files for other languages.

Katsuyuki-Karasawa, 2022-08-15:
---
name: i18n Roadmap
about: i18n support and associated language additions
title: #719

---

## Usage

- Close Issue when all work is complete
-  If the work is unnecessary, do not check the box.

## Language Coverage

- [ ] Top page
- [ ] Keymaps Configurator
- [ ] Catalog Page
- [ ] Keyboard Registration
- [ ] etc...

## i18n compatible method
### Library
- [ ] [i18next](https://www.i18next.com/)
- [ ] etc...

### Implement by yourself
- [ ] Consider your own means of implementation.

## Where language files should be created or changed

- [ ] `.json`
- [ ] [crowdin](https://crowdin.com/)
- [ ] Write directly in .js(?)
- [ ] etc...

## Target language
- [ ] ja-JP(Most users & first priority)
- [ ] en-US(begin one's text once)
- [ ] en-UK(?)
- [ ] ko-KR(Custom keyboards, in terms of homemade keyboards
Also, I think there are more users in Asia next to Japan)
- [ ] zh-CN
- [ ] zh-tw(I feel like simplified only is fine, but in case it's not in the language code, it's not safe to handle files without it.)

### Checkboxes are set to `true` as soon as the translation for each language is completely finished

yoichiro, 2022-08-23:
@Katsuyuki-Karasawa Thank you for the preparation. There is the "i18next" in the Library section. Do we also need the "react-i18next"? I agree we use the "i18next", but it would be better that we use the "react-i18next", too. How about this?

yoichiro, 2022-08-24:
@Katsuyuki-Karasawa Also, I think that we need to use the `i18next-browser-languagedetector` library to detect a language. And, I guess that the priority list of detection methods is `'[ querystring', 'cookie',  'navigator', 'localStorage', 'htmlTag ]'`. How about them?

yoichiro, 2022-08-24:
@Katsuyuki-Karasawa If you agree two my comments above, I intend to create a prototype. I think that the prototype supports two languages (English and Japanese) on only the top page using the libraries (i18next and so on). Anyway, could you confirm the comments above and reply to them? Thank you in advance!

yoichiro, 2022-09-19:
@Katsuyuki-Karasawa Hello? Could you answer my questions above?

yoichiro, 2022-09-21:
I finally were not able to get any answers from @Katsuyuki-Karasawa. Therefore, I close this issue.

closed 2022-09-21 by yoichiro