remap-keys/remap #737: Claw44 `rev2` can't apply shared keymaps

gshmu, 2022-09-25:
my claw44 no shared keymaps
[image]

the shared keymaps list
[image]

please help

yoichiro, 2022-10-19:
claw44 rev2 is not the same as Claw44.

[image]

closed 2022-10-19 by yoichiro

gshmu, 2022-11-04:
claw44 rev2 is based claw44, (add some new keys), I want apply claw keymap to rev2.

yoichiro, 2022-11-04:
I know that, but the VID and PID of the claw44 rev 2 is different from them of claw44. Therefore, they are different keyboard in Remap.

gshmu, 2022-11-11:
@yfuku can update the VIP and PID?

yfuku, 2022-11-14:
@gshmu 
Thank you for using Claw44.
Claw44 rev1 and rev2 have different wiring, so they cannot share the same keymap.
We apologize for the inconvenience, but we would like to ask you to start from the beginning.