remap-keys/remap #745: Yarn Build error

aihk, 2022-10-03:
Creating an optimized production build...

<--- Last few GCs --->

[1204:00000001009E0360]   305777 ms: Scavenge 1008.6 (1037.0) -> 1008.2 (1045.0)
 MB, 29.2 / 0.0 ms  (average mu = 0.139, current mu = 0.015) allocation failure

[1204:00000001009E0360]   307254 ms: Mark-sweep 1008.7 (1045.0) -> 1007.4 (1044.
6) MB, 1474.1 / 0.1 ms  (average mu = 0.102, current mu = 0.047) allocation fail
ure scavenge might not succeed

<--- JS stacktrace --->

FATAL ERROR: Ineffective mark-compacts near heap limit Allocation failed - JavaS
cript heap out of memory
 1: 00007FF71C14B1EF v8::internal::CodeObjectRegistry::~CodeObjectRegistry+12359
9
 2: 00007FF71C0D8BA6 v8::internal::MicrotaskQueue::GetMicrotasksScopeDepth+65206

 3: 00007FF71C0D9C7D node::OnFatalError+301
 4: 00007FF71CA0C20E v8::Isolate::ReportExternalAllocationLimitReached+94
 5: 00007FF71C9F67ED v8::SharedArrayBuffer::Externalize+781
 6: 00007FF71C899DCC v8::internal::Heap::EphemeronKeyWriteBarrierFromCode+1468
 7: 00007FF71C8A6A79 v8::internal::Heap::PublishPendingAllocations+1129
 8: 00007FF71C8A3A4A v8::internal::Heap::PageFlagsAreConsistent+2842
 9: 00007FF71C8966A9 v8::internal::Heap::CollectGarbage+2137
10: 00007FF71C89EF6B v8::internal::Heap::GlobalSizeOfObjects+331
11: 00007FF71C8E56DB v8::internal::StackGuard::HandleInterrupts+891
12: 00007FF71C608AA2 v8::internal::RegExpParser::simple+1410
13: 00007FF71C393C0B v8::internal::RegExpMacroAssemblerX64::CheckStackGuardState
+59
14: 0000000106A305DB

aihk, 2022-10-03:
> Remap@0.2.3 build
> react-scripts build

Creating an optimized production build...
Compiled successfully.

File sizes after gzip:

  1.17 MB    build\static\js\2.8ca0e278.chunk.js
  192.81 KB  build\static\js\main.f757388c.chunk.js
  9.87 KB    build\static\css\main.f04f987c.chunk.css
  1.66 KB    build\static\css\2.3db94de3.chunk.css
  1.62 KB    build\static\js\3.957206b4.chunk.js
  1.16 KB    build\static\js\runtime-main.8feed9c1.js

The bundle size is significantly larger than recommended.
Consider reducing it with code splitting: https://goo.gl/9VhYWB
You can also analyze the project dependencies: https://goo.gl/LeUzfb

The project was built assuming it is hosted at /.
You can control this with the homepage field in your package.json.

The build folder is ready to be deployed.
You may serve it with a static server:

  yarn global add serve
  serve -s build

Find out more about deployment here:

  https://cra.link/deployment

yoichiro, 2022-10-19:
@aihk Could you report the steps to reproduce this issue? I don't have any experience about the error.

yoichiro, 2022-10-25:
I was not able to get enough information from the reporter. I close this issue.

closed 2022-10-25 by yoichiro