remap-keys/remap #751: The layered function keys are assigned to different keys in version 0.19.2 of QMK.

Salicylic-acid3, 2022-12-06:
最新のQMK環境で書き込んだ場合、レイヤー機能キー（例えばMO）が違うキーとして認識されるようです。
複数のキーボードのファームウェアで確認しました。
ちなみにMO(1)でCtrl、MO(2)でShiftでした。

When writing in the latest QMK environment, it seems that the layer function keys (e.g. MO) are recognized as different keys (Ctrl for MO).
I have confirmed this with multiple keyboard firmwares.
(By the way, it was Ctrl for MO(1) and Shift for MO(2).

yswallow, 2022-12-06:
Since [QMK Breaking Changes on Nov 26, 2022](https://github.com/qmk/qmk_firmware/blob/master/docs/ChangeLog/20221126.md#keycodes-refactoring-idkeycodes-overhaul-core-changes), keycodes except basic keycodes are changed.

It is not only the layer keys, but also RGB keys and Mouse keys.

The information of all the keys is here. https://keyboards.qmk.fm/v1/constants/keycodes_0.0.1.json

I think the keycodes version can be recognized by using `get_protocol_version`: `0x01` in RAW HID. If the response is `0x000A` or below, send the keycodes which are used in Remap currently. If that is `0x000B`, we need to send the latest keycodes. https://github.com/qmk/qmk_firmware/blob/4020674163fc80914059c4c9c3be5c0ae00bd150/quantum/via.h#L61

yoichiro, 2022-12-07:
@yswallow @Salicylic-acid3 Thank you for the reporting. Well, this change is so large impact against Remap. Probably, Remap can't switch a key code structure depending on the VIA protocol version, because there is no matter between the VIA protocol version and the QMK revision, I guess. Anyway, I intend to understand the detail of the change as the first approach.

yoichiro, 2022-12-07:
@yswallow I have one question. Do you know how to generate the [keycodes_0.0.1.json](https://keyboards.qmk.fm/v1/constants/keycodes_0.0.1.json)?

yswallow, 2022-12-07:
I don’t know. I found this URL on [QMK Documentation](https://github.com/qmk/qmk_firmware/blob/master/docs/api_docs.md#constants-idqmk-constants).

JohnAZoidberg, 2023-02-01:
@yoichiro 
> @yswallow I have one question. Do you know how to generate the [keycodes_0.0.1.json](https://keyboards.qmk.fm/v1/constants/keycodes_0.0.1.json)?

It's all of the files here: https://github.com/qmk/qmk_firmware/tree/master/data/constants/keycodes
merged together.

yoichiro, 2023-09-12:
The new version of Remap has been released on Sep 11th, 2023. The version is supporintg QMK Firmware 0.22.2 or higher and VIA Protocol version 0x0C.

closed 2023-09-12 by yoichiro