remap-keys/remap #771: Unregistered Firmware source location keyboards are not editable

yfuku, 2023-10-30:
I attempted to register my keyboard definition from an individual to an organization. While most modifications were successful, I encounter an error when trying to modify a keyboard definition with an 'Unknown' Firmware source location.

Steps to reproduce:

Select the keyboard from the REGISTER/MANAGE KEYBOARDS page.
Click on the 'BACK TO DRAFT' link under the DEFINITION tab.
An error message 'Updating the Keyboard Definition failed' is displayed.
The page remains stuck on 'Processing...'
This issue pertains to two keyboards: bat43 and Claw44 rev1.
It is likely that these keyboards were registered during an earlier period.
[image]

yoichiro, 2023-11-04:
@yfuku Thank you for submitting this issue and talking in person today. I intend to confirm this issue and will fix this as soon as possible. Probably I will try to fix this issue next week.

yoichiro, 2023-11-14:
@yfuku This issue should be fixed at #775. Could you confirm it? You can reopen this issue if this issue is not solved.

yfuku, 2023-11-17:
@yoichiro 
Thank you for addressing this issue.
An error has occurred again. It seems that there is no data in the database for qmkRepositoryFirstPullRequestUrl.

Upon comparing with keyboards that are registered correctly, it was found that qmkRepositoryFirstPullRequestUrl and contactInformation are missing. However, since contactInformation is defined as 'string | undefined', I believe the latter should not be a problem.

I'm sorry, but I can't make a pull request because I am unable to replicate the admin panel.

yfuku, 2023-11-17:
It seems that only collaborators or the person who closed the issue can reopen it. Sorry, I can't change the status