remap-keys/remap #883: character highlight in selection menu isn't initially displayed

mxmilkiib, 2025-04-21:
this is me  bringing the list up, then pressing down, then up

[image]

[image]

[image]

pressing enter just after opening the menu to select the first item works

it's just not highlighted at that point

AbiodunVlad, 2025-06-24:
@mxmilkiib I don't really understand what you say. Can you explain what the expected behaviour is? Then clearly explain what you experience.
Thank you.

mxmilkiib, 2025-06-24:
in the drop-down menu, there is a selected item that is "highlighted" by using a darker background colour, e.g. how most drop-downs work 

one can move that selection up and down using the up and down arrows

the problem is that that bg highlight doesn't appear for the first item when the drop-down menu is first opened 

this is just appearance though, the first item is actually selected

pressing the down arrow moves the selected menu item down one, that then finally displays the darker selection bg, on the second item

from then one can visually see the selection as that dark bg goes up n down on arrow presses

what should happen; when the menu opens, the first item should have a darker bg

AbiodunVlad, 2025-06-24:
After the issue has been properly broken down and explained, I now understand and I want to work on it. It requires CSS to understand how to interact with the first child element.